adonno/tagreader #167: Can´t read NFC tags

nedflanders11, 2022-06-01:
Hello!

Thanks for this wonderful project.

I´ve have received my tagreader this week. It worked fine the first day (integrated in my Home Assistent, reading tags, automations, ...). It was perfect. 

In the evening I have unplugged the USB-C. On the next day I have plugged in the USB-C again. After a few seconds the tagreader was in status "connected" (Home Assistant).

From now on the tagreader didn´t read tags anymore. It´s still possible to switch on/off the LED, restarting, ..., but reading tags is not possible.

In the esphome-flasher is no log after scanning. But there are log entries when I switch on/off the LED, ...

Any suggestions? Thanks!

adonno, 2022-06-01:
Can you update your tagreader via[https://adonno.github.io/tagreader]() and let me know?

nedflanders11, 2022-06-01:
Thanks for your fast feedback. 

Recommended update (ESPHome 2022.3.2) successfully done.
Unfortunately still the same behavior. 
I have deleted the ESPHome integration in Home Assistant. Afterwards integrated again. Same behavior.

nedflanders11, 2022-06-01:
I´ve done a restart and have seen the following three errors:

I][logger:214]: Log initialized
[C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[I][i2c.arduino:161]: Performing I2C bus recovery
**[E][i2c.arduino:177]: Recovery failed: SCL is held LOW on the I2C bus**
[D][template.switch:050]:   Restored state ON
[D][switch:013]: 'TagReader Buzzer Enabled' Turning ON.
[D][switch:037]: 'TagReader Buzzer Enabled': Sending state ON
[D][template.switch:050]:   Restored state ON
[D][switch:013]: 'TagReader LED enabled' Turning ON.
[D][switch:037]: 'TagReader LED enabled': Sending state ON
[C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[C][light:036]: Setting up light 'TagReader LED'...
[D][light:035]: 'TagReader LED' Setting:
[D][light:040]:   Color mode: RGB
[D][light:084]:   Transition length: 1.0s
[C][pn532:018]: Setting up PN532...
**[E][pn532:022]: Error sending version command
[E][component:112]: Component pn532 was marked as failed.**

adonno, 2022-06-02:
Would you mind opening the device and sending me an image of the soldering spots please ?

nedflanders11, 2022-06-03:
Please see attachment:
[image]

adonno, 2022-06-03:
I'll send you a new one i guess , 
Can you email me your initial order number  adonno@adonno.com

adonno, 2022-06-05:
Ok i will come back to it by end of week

adonno, 2022-11-01:
sent a new device

closed 2022-11-01 by adonno