adonno/tagreader #168: diffent power supply

Giuliano69, 2022-06-02:
Now the ONLY  supply source for the kit is a usb power adapter.

it could be VERY  helpful if the package could be powered in a easy way in a domotic installation.
Typically domotic installation includes : 
-POE
-knx power line  (30V DC)

HOW can we manage to easily integrate the power supply of the D1 with the said power sources ?
Can it be offered ALSO in such common variant (eg knx) ?

BR

Giuliano69, 2022-06-03:
Seems that only esp32 chip could use ethernet, and then have POE.
https://esphome.io/components/ethernet.html

adonno, 2022-06-05:
There is an esp32with poe version existing https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware 
for the rest you would need a buck convertor to step down from 30V to 5V 

The devices I build are build in a small Form factor and i'm trying to make them as small as possible.

Giuliano69, 2022-06-05:
Thanks for our reply.

For what appears from the esp32 & es8266 comparison , the ethernet module
is NOT present in the esp8266.
So the POE  must be coupled with an esp32 version.
https://www.electronicshub.org/esp32-vs-esp8266/

For a domestic installation, a commercial power bus capable of powering the
module, would be a great option.

BR
Giuliano

adonno, 2022-06-05:
I have KNX myself but never considered making a KNX version one

jgelderloos, 2022-06-13:
Hey, I am also interested in a POE version of this and may trying building it out myself. @adonno are you willing to share the CAD files for the case so I can attempt making another version to fit what is needed?

I think it may be simpler to just continue with the esp8266 and only use the ethernet for power supply.

adonno, 2022-11-01:
I prefer not to share the original files they are a mess that's why I shared the STLs

closed 2022-11-01 by adonno