adonno/tagreader #169: Tagreader is not discoverd by ESPHome dashboard

arjenfvellinga, 2022-06-06:
Summary
I'm running Home Assistant (2022.5.5) and ESPHome dashboard (2022.5.1) isn't discovering my tagreader

I had Home Assistant running without the ESPHome add-on installed. Then I added a tag reader to my setup. I can't remember if it was discovered by Home Assistant or that I added it manually. But the result was a correctly configured tagreader as a ESPHome integration.

Recently, I purchased some smart plugs based on ESPHome and added the ESPHome add-on including the dashboard. The dashboard discoverd the smart plugs, but it didn't find the tagreader

Integration:
[image]

add-on
[image]

How can I have the dashboard also list the tagreader?

riemers, 2022-07-24:
fyi, just configured mine, i don't see it either so might be "business as usual"

adonno, 2022-07-24:
When did you get yours ? Or did you DIY it ?

riemers, 2022-07-24:
> When did you get yours ? Or did you DIY it ?

Got mine at least multiple months ago. Found it again on the pile of “I want to use it but didn’t have time for it yet” 🤔 does have usb-c though.

adonno, 2022-11-01:
oh wow sorry i missed replying to this thread 

can you guys flash it via https://adonno.github.io/tagreader

arjenfvellinga, 2022-11-02:
I indeed flashed it, and it was found. So for me it works!

closed 2022-12-10 by adonno