adonno/tagreader #173: Tag Reader Pingable, but cannot connect

rlburkey, 2022-07-20:
Good Afternoon!

After a wifi outage my tag reader is no longer functioning properly.  After wifi was restored here is the current state:

The tag reader has a solid red status light
The tag reader is obtaining an IP address
The tag reader is pingable by that IP address
The web interface for the tag reader is not working
When a fob/card (both previously worked) is tapped on the tag reader, the light does not change and there is no longer a buzz/beep.

I have power-cycled the tag reader, no change.  Please let me know if there is a reset procedure or what other steps I can take to restore service.

Thanks!

riemers, 2022-07-24:
by default, the buzzer/light is disable in home assistant as far as i know. if you added it in HA (new device found/integrations) it should create those switches for you and you can use that. (just trying to help out, was playing with it right now too)

adonno, 2022-07-24:
Maybe try reflashing it. Go to https://adonno.github.io/tagrader

And try reinstalling tagreader.

rlburkey, 2022-07-31:
I tried reflashing the reader.  When using the tagreader.yaml file it does not create its own network and does not connect to the any network I have setup if I configure it in the yaml file.

If I install the generic ESPHome firmware, it does connect to the network successfully.

adonno, 2022-07-31:
> I tried reflashing the reader.  When using the tagreader.yaml file it does not create its own network and does not connect to the any network I have setup if I configure it in the yaml file.
> 
> 
> 
> If I install the generic ESPHome firmware, it does connect to the network successfully.
> 
> 

what yaml? the one from this repo? 
or the one from https://adonno.github.io/tagreader?

adonno, 2022-11-01:
no reply for 3 months

closed 2022-11-01 by adonno