adonno/tagreader #183: Have i bought the wrong rfid cards?

steinis123, 2022-08-24:
I have cards that work fine on my tag reader so I bought some more, but i cant get them to work.

The card is of type: 125Khz Card ID token tags RFID (Aliaxpress name:P).

Is this the wrong type of card, if so what should i look after when i buy som more?:)

PaulHam211, 2022-08-27:
eBay title for the cards I'm using NTAG215 Blank NFC Cards PVC Tags NTAG 215 TagMo Amiibo Waterproof RFID

JeeCee1, 2022-11-16:
I'd like to add that NTAG 213/215/216 only differ in storage space; 144/504/888 Bytes respectively.

"RFID" defines 3 frequency spectra: 125-134 kHz, 13.56 MHz and 856-960 MHz.
NFC is a subset of the generic RFID, and only works on 13.56 MHz.
All NTAG cards (should) work on 13.56 MHz.
See also [here](https://www.asiarfid.com/de/difference-ntag213-ntag215-ntag216.html) (German) and [here](https://www.atlasrfidstore.com/rfid-insider/rfid-vs-nfc/).

Since the reader only supports [NFC](https://www.nxp.com/docs/en/nxp/data-sheets/PN532_C1.pdf) (13.56 MHz), unfortunately your cards are unsuitable.

@adonno Maybe good info for the frontpage?