adonno/tagreader #184: Mute buzzer upon restart or lost wifi

steinis123, 2022-08-30:
Is it possble to add a function to mute the buzzer when the unit is restarting or power up? I have one at my kids room and some times the unit restart due to lost wifi connection.

tempod, 2022-09-20:
you have to insert the hash in front of the line

`- rtttl.play: "success: d = 24, o = 5, b = 100: c, g, b"`

in the section

`on boot:`

closed 2022-11-01 by adonno

foothook, 2023-11-21:
> you have to insert the hash in front of the line
> 
> `- rtttl.play: "success: d = 24, o = 5, b = 100: c, g, b"`
> 
> in the section
> 
> `on boot:`

Hi i try to obtain this, mute on boot -  but i am quiet new to ESPhome and ESP in general.
I have the prebuild reader from adonno - adopted in my esphome but in my yaml I have only the code for wifi and api key, the rest is imported trough
`packages:
  adonno.tag_reader: github://adonno/tagreader/tagreader.yaml`

may I ask for your help, how can i modify the yaml to change this part of the code?