adonno/tagreader #187: tagreader not reading tags?

copywizard, 2022-09-10:
Hello 

Firstly thanks a lot for this nice project.

I have ordered all parts from aliexpress and just got them in and soldered them together using your schematics.
But somehow the device won,t scan NFC carts and the led will also not work.

The ESP connects to my HA instance perfectly so i have some logs 
[logs_tagreader-e83bc6_logs.txt](https://github.com/adonno/tagreader/files/9540086/logs_tagreader-e83bc6_logs.txt)

Could you please help me out here?

I have created a NFC tag on the included cart that came with the NFC PCB i wrote a automation/link to it using my mobile phone and using my mobile phone it works fine but for some reason the tagreader won,t read it at all.

adonno, 2022-09-12:
Can you please provide pictures of the setup ?

pvh0, 2022-09-12:
I think you forgot to change the first dip switch on the nfc board to on?

pvh0, 2022-09-12:
And I think you got the soldering wrong, nfc bord is not connected to 5v?

copywizard, 2022-09-12:
you where right it was multiple factors like me not reading enough and the dip switch was overlooked and some soldering was reversed :( thanks for the help it seems to work correctly now this can be closed

pvh0, 2022-09-12:
Good to hear it helped. I’m jealous since I was also struggling today to get it working on esp32 but no luck yet!

abdullahvenom21, 2024-07-12:
I also have a similar problem i connected my wiring correct and the switch is set up correctly but my device is not reading tags and i tried with card that came with the tag module and i also tried with credit cards and it is not working can some one plz help me @adonno @pvh0 
[image]
[image]