adonno/tagreader #189: disable automatic tag addition

tempod, 2022-09-13:
is it possible to disable the automatic addition of tags to home assistant?
it would be great to have a switch to enable and disable the auto added

tempod, 2022-09-20:
i tried to do it myself but i need help.
I added a switch for auto tag addition and a condition. but it doesn't seem to work.
where did I go wrong?
here is the code:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.160
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "TagReader Fallback Hotspot"
    password: !secret wifi_password_fallback

captive_portal:

web_server:
  port: 80

improv_serial:

substitutions:
  name: tagreadergarage1
  friendly_name: TagReader

esphome:
  name: $name
  platform: ESP8266
  board: d1_mini

  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  # name_add_mac_suffix: true

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: adonno.tag_reader
    version: "1.4"
# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
#     - rtttl.play: "success:d=24,o=5,b=100:c,g,b"
#    - light.turn_on:
#        id: activity_led
#        brightness: 100%
#        red: 0%
#        green: 0%
#        blue: 100%
#        flash_length: 500ms
    - switch.turn_on: buzzer_enabled
    - switch.turn_on: led_enabled

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true
  restore_state: true
  entity_category: config
[... 307 lines cut ...]
    ip_address:
      name: "${friendly_name} IP Address"
      icon: mdi:wifi
      entity_category: diagnostic
    ssid:
      name: "${friendly_name} Connected SSID"
      icon: mdi:wifi-strength-2
      entity_category: diagnostic

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

# Configure LED
light:
- platform: neopixelbus
  variant: WS2812
  pin: D8
  num_leds: 1
  flash_transition_length: 500ms
  type: GRB
  id: activity_led
  name: "${friendly_name} LED"
  restore_mode: ALWAYS_OFF
```

adonno, 2022-11-01:
will have a look later today and come back with a solution

adonno, 2022-11-03:
this hwole part need to be indented further 

```
    - if:
        condition:
          switch.is_on: addtag_enabled
        then:        
        - lambda: |-
            id(source)="";
            id(url)="";
            id(info)="";
            if (tag.has_ndef_message()) {
              auto message = tag.get_ndef_message();
              auto records = message->get_records();
              for (auto &record : records) {
                std::string payload = record->get_payload();
                std::string type = record->get_type();
                size_t hass = payload.find("https://www.home-assistant.io/tag/");
                size_t applemusic = payload.find("https://music.apple.com");
                size_t spotify = payload.find("https://open.spotify.com");
                size_t sonos = payload.find("sonos-2://");
                if (type == "U" and hass != std::string::npos ) {
                  ESP_LOGD("tagreader", "Found Home Assistant tag NDEF");
                  id(source)="hass";
                  id(url)=payload;
                  id(info)=payload.substr(hass + 34);
                }
                else if (type == "U" and applemusic != std::string::npos ) {
                  ESP_LOGD("tagreader", "Found Apple Music tag NDEF");
                  id(source)="amusic";
                  id(url)=payload;
                }
                else if (type == "U" and spotify != std::string::npos ) {
                  ESP_LOGD("tagreader", "Found Spotify tag NDEF");
                  id(source)="spotify";
                  id(url)=payload;
                }
                else if (type == "U" and sonos != std::string::npos ) {
                  ESP_LOGD("tagreader", "Found Sonos app tag NDEF");
                  id(source)="sonos";
                  id(url)=payload;
                }
                else if (type == "T" ) {
                  ESP_LOGD("tagreader", "Found music info tag NDEF");
                  id(info)=payload;
                }
                else if ( id(source)=="" ) {
                  id(source)="uid";
                }
              }
            }
            else {
              id(source)="uid";
            }
        
        - if:
            condition:
              lambda: 'return ( id(source)=="uid" );'
            then:
              - homeassistant.tag_scanned: !lambda |-
                  ESP_LOGD("tagreader", "No HA NDEF, using UID");
                  return x;
            else:
            - if:
                condition:
                  lambda: 'return ( id(source)=="hass" );'
                then:
                - homeassistant.tag_scanned: !lambda 'return id(info);'
                else:
                - homeassistant.event:
                    event: esphome.music_tag
                    data:
                      reader: !lambda |-
                        return App.get_name().c_str();
                      source: !lambda |-
                        return id(source);
                      url: !lambda |-
                        return id(url);
                      info: !lambda |-
                        return id(info);
    
    - if:
        condition:
          switch.is_on: buzzer_enabled
        then:
        - rtttl.play: "success:d=24,o=5,b=100:c,g,b"

```

[image]

closed 2022-11-03 by adonno

tempod, 2023-01-01:
unfortunately only now I was able to do this test.
I noticed that it only reads tags if I turn on the add tag switch. if deactivated, it does not even read those already memorized