adonno/tagreader #191: What do these options do?

bmbouter, 2022-09-29:
Can a description be given for what these do?

* Write a tag random
* Clean Tag
* Cancel Writing

Also for the configuration section, are the LED enabled and Buzzer Enabled for the device after booted, or are these the states of the device when it boots? How can I control the state when it boots, is that only in the yaml itself?

bmbouter, 2022-09-29:
Also what's the upgrade like, when a new version is released does esphome auto-upload it and apply the substitutions?

adonno, 2022-11-01:
- Write a tag at random, enabling tis issue it will write a random url to the tag that will work with HA. 
- Clean tag, will erase the tag and all of its contents 
- Cancel writing will cancel above operations

Settings for LED and Buzzer will have to be added to the yaml via esphome and will overwrite the default settings 

> Also what's the upgrade like, when a new version is released does esphome auto-upload it and apply the substitutions?

no you will have to manually upload it but esphome will find an update when changes are made to the main yaml and you will get an update notification

closed 2022-11-01 by adonno