adonno/tagreader #193: Homeassistant read ANY tag

IAmTheMilkManMyMilkIsDelicious, 2022-10-05:
This is probably more of a homeassistant question than the tagreader itself. But does anyone know how I could create an automation that is triggered by reading any tag?

I can make a trigger in my automation for a specific tag such as:
```
platform: tag
tag_id: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
```

But what I'd like to do is a trigger on any tag (even unknown tags). Then a condition will check if the scanned tag is equal to a specific tag and do an action, otherwise send me a notification that an unknown tag was scanned.

Has anyone implemented something like this?

luka6000, 2022-10-14:
you can do this by triggering homeassistant.event service instead of homeassistant.tag_scanned. Look at the code how it’s done for music_tags and just modify it for uid scanned tags as needed.
This would be breaking change for adonno reader so it’s probably not going to happen here.

Mindsgoneawol, 2022-10-19:
I had tags triggering automationsto play movies on roku media player from jellyfin integration up until I updated to this last esphome version. Can't seem to fix this issue. The only reason I made these scanners.

luka6000, 2022-10-19:
> I had tags triggering automationsto play movies on roku media player from jellyfin integration up until I updated to this last esphome version. Can't seem to fix this issue. (was fun while it lasted lol)

This issue is a feature request so this never worked in tagreader. Your video automations probably can be switched to tags scanned with tagreader and will work just fine.

Mindsgoneawol, 2022-10-19:
The tags no longer show up in home assistant at all when I scan them. They did work. I made the mistake of updating esphome. Now the tags scan and show up in the esphome logs but home assistant isn't seeing them. I have a video of it starting an automation for plex to play a movie. (thats hit or miss .. worked better with jellyfin) If you want to see it I will send it to you. I assure you it did work. Native feature would be awesome though. Tried others configurations flashed to this and they don't work so i don't think its this configuration. My guess its esphome or my HA is messed up.

adonno, 2022-11-01:
not sure i'm following, what is it you want to do exactly since at first  you want to trigger an automation based on any tag 
and then you talks about movies playing issues ?

IAmTheMilkManMyMilkIsDelicious, 2022-11-01:
I was the original poster and I have solved my issue which was creating an automation that can read any tag. Here is an example of the automation yaml:

```
alias: Tag Scan
description: ""
trigger:
  - platform: event
    event_type: tag_scanned
condition: []
action:
  - choose:
      - conditions:
          - condition: template
            value_template: >-
              {% set data = trigger.event.data %} {{ data.tag_id ==
              "xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx" }}
        sequence:
              - service: script.xxx_tag_scan_script
                data: {}
      - conditions:
          - condition: template
            value_template: >-
              {% set data = trigger.event.data %} {{ data.tag_id ==
              "yyyyyyyy-yyyy-yyyy-yyyy-yyyyyyyyyyyy" }}
        sequence:
              - service: script.yyy_tag_scan_script
                data: {}
    default:
      - service: script.notify_invalid_tag
        data_template:
          input_text: |-
            An unknown NFC key was used!
            ID: {{ trigger.event.data.tag_id }}
mode: single

```

closed 2022-11-01 by IAmTheMilkManMyMilkIsDelicious