adonno/tagreader #194: Not able to write music Tag

gbfromhb, 2022-10-07:
I'm able to read a card just fine. It writes the tag ID and creats a tag in Home Assistant. I have created an Automation using this, and I'm able to play a Spotify Album to a Google Cast device.

What I'm having issues with is writing a music tag. 

I first use the "Clean Tag" button. Your tag reader beeps, and the LED turns yellow.

I wave the card over the reader/writer, it beeps and the yellow light goes off.

I then call the "ESPHome: family_room_tag_reader_write_music_tag" service with the following.
music_url: 
   spotify:album:7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-WqSOzWF1b_eA7vEJAtP3KgKSpOHVgwm3Eh

music_info: Beatles - 1

The tag reader beeps and the LED turns red.

I wave the card over the reader/writer, it beeps and the red light goes out.

I read the card, and this is what is being read from the card

event_type: tag_scanned
data:
  tag_id: 04-72-90-BA-AD-6C-81
  device_id: ad572ebe52e0a4eccf7726abadeeb8ca
origin: LOCAL
time_fired: "2022-10-07T22:41:56.383861+00:00"
context:
  id: 01GET9SH4ZNN8TDYH4FXV3HQQ8
  parent_id: null
  user_id: null

Any suggestion on what I may be doing wrong.

hannemann, 2022-10-08:
I am also playing around with this atm.
You have to prefix your URL with `https://open.spotify.com/` and listen to the event `esphome.music_tag`. You can find valid prefixes in the yaml. Search for `open.spotify` and you will see...

I also added my own URL prefix lms:// since i want to send the data being read to my Logitech Media Server.

gbfromhb, 2022-10-08:
I'm using the latest yaml file, and I don't seem to have the esphome.music_tag event avaliable either. I have recompile and had Home Assistant reinstall the tagreader device and the event does not seem to be there.

I have also installed the blueprint and the tagreader does not appear as a choice. I'm not sure where to go with this one.

hannemann, 2022-10-09:
Did you rewrite the URL? It should look like `https://open.spotify.com/album/7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-`. If not the `esphome.music_tag` event is not fired but `tag_scanned` instead.

One of the prefixes 
- https://open.spotify.com
- https://music.apple.com
- sonos-2://
is mandatory for the `esphome.music_tag` event to be fired.

AFAIK the event becomes visible in HA as soon as you use it in an automation or used it at least once... not sure.
Try to listen to the event anyway. It should work even if it does not show up in the list.
  
I have ordered cards that are rectangular and slightly wider (45mm x 24mm) than the reader module (the red pcb) and smaller round ones that have 20mm in diameter. The rectangular ones work much better. Even through the 2.5 cm wooden tabletop! (the reader is screwed to the underside of the table facing down)

The round ones work reliably most of the time if i place them on the edge of the reader module instead of the center. I think that's related to the antennae which i think is the white rectangle on pcb but i could be wrong about that. If i place them in the center of the scan area than i mostly get the same results like you mentioned. Maybe they don't get enough energy or there is an interference or...

With both types of cards it works best if i hold them still a centimeter or two above the scan area for a few seconds. Try it and observe the esphome log of your reader.
You should see something like:
```
[02:44:08][D][pn532:162]: Found new tag '04-86-BA-06-19-6C-80'
[02:44:08][D][pn532:166]:   NDEF formatted records:
[02:44:08][D][pn532:168]:     U - lms://Radio Stationen/0
[02:44:08][D][pn532:168]:     T - Radio Paradise
[02:44:08][D][pn532:295]: Waiting to read next tag
```
The U - should be `https://open.spotify.com/album/7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-` in your case.
If you see something like:
```
[02:46:32][D][pn532:162]: Found new tag '04-34-29-6C-39-61-80'
[02:46:32][D][pn532:295]: Waiting to read next tag
```
the reading failed but the `tag_scanned` event is fired anyway with the uid of the tag instead of an `esphome.music_tag` event.

Hope that helps a bit.

gbfromhb, 2022-10-10:
I think I have it working, but the problem I'm having is that I'm pretty sure for me to get Spot Cast to work with my configuration is to use the "spotify:" not the "https://open.spotify.com/". Is there anyway you can also trigger the esphome.music_tag with this prefix as well?

hannemann, 2022-10-11:
It's possible but you have to alter the code...

Why not transform it in your automation?

If your url would look like `https://open.spotify.com/album/7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-` you could transform it with Jinja within HA like so (Developer Tools -> Template):
```
{% set parts = "https://open.spotify.com/album/7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-".replace('https://open.spotify.com/', '').split('/') %}
{{ ['spotify', parts[0], parts[1]] | join(':') }}
```
I don't have a clue how spot cast works but within an automation you could use the trigger data:
```
service: media_player.play_media
target:
  entity_id: media_player.the_player
data:
  media_content_type: music
  media_content_id: >
    {% set parts = trigger.event.data.url.replace('https://open.spotify.com/', '').split('/') %}
    {{ ['spotify', parts[0], parts[1]] | join(':') }}
```
Something like that.

gbfromhb, 2022-10-11:
That was the direction I was heading, and thank you for the code. The prefix I'm using has to do with Home Assistant (I'm pretty sure). This seems to be the only way I'm able to play Spotify via a service or automation.

gbfromhb, 2022-10-11:
First of all thank you for everthing and your already assembled tag reader the best. I ended up modifiing your yaml file and changed this part. I added the spotify_2 variable and delt with it in the conditionals. Here is the section I modified.

`    - lambda: |-
        id(source)="";
        id(url)="";
        id(info)="";
        if (tag.has_ndef_message()) {
          auto message = tag.get_ndef_message();
          auto records = message->get_records();
          for (auto &record : records) {
            std::string payload = record->get_payload();
            std::string type = record->get_type();
            size_t hass = payload.find("https://www.home-assistant.io/tag/");
            size_t applemusic = payload.find("https://music.apple.com");
            size_t spotify = payload.find("https://open.spotify.com");
            size_t spotify_2 = payload.find("spotify:");
            size_t sonos = payload.find("sonos-2://");

            if (type == "U" and hass != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Home Assistant tag NDEF");
              id(source)="hass";
              id(url)=payload;
              id(info)=payload.substr(hass + 34);
            }
            else if (type == "U" and applemusic != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Apple Music tag NDEF");
              id(source)="amusic";
              id(url)=payload;
            }
            else if (type == "U" and spotify != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Spotify tag NDEF");
              id(source)="spotify";
              id(url)=payload;
            }
            else if (type == "U" and spotify_2 != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Spotify tag NDEF");
              id(source)="spotify";
              id(url)=payload;
            }
            else if (type == "U" and sonos != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Sonos app tag NDEF");
              id(source)="sonos";
              id(url)=payload;
            }
            else if (type == "T" ) {
              ESP_LOGD("tagreader", "Found music info tag NDEF");
              id(info)=payload;
            }
            else if ( id(source)=="" ) {
              id(source)="uid";
            }
          }
        }
        else {
          id(source)="uid";
        }
`

hannemann, 2022-10-12:
I'm not the author of the code nor did i develop the tagreader hardware ;) I stumpled upon your issue because i had a similar problem ;)

I would not recommend to modify the tagreader code since you have to reapply your changes in case of an update. If you could do it in HA that's the preferred way, easier and more flexible.

If you need that string: `spotify:album:7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-` program your tag with the url `https://open.spotify.com/spotify:album:7vEJAtP3KgKSpOHVgwm3Eh?si=N6nfodcgS-` and use it in HA like so:
```
"{{ trigger.event.data.url.replace('https://open.spotify.com/', '') }}"
```

adonno, 2022-11-01:
this will wrok fine, even when reflashing or updating the tagreader @gbfromhb 's modification should remain untouched. 

The tagreader.yaml is a community effort :-D not only mine 
the PCB is also a community effort :-D

closed 2022-11-01 by adonno