adonno/tagreader #203: Can't get tag read

dyay108, 2022-11-14:
I've tried with 3 different esp8266 modules and 3 different pn532 boards and can't get it working.

I've triple checked the wiring, soldering, etc...
Home assistant running locally and not exposed to the Internet.

[Link to esp8266](https://a.co/d/9eArYjs) 

[image]

Successfully added the module to hass.

[image]

adonno, 2022-11-14:
This is wierd it seems to find the PN532 properly though did you change the wires ? (i know this might be a dumb question. 
Can you try soldering the wire directly to the pins ?  

Maybe the cards are defective ? 
Tag reader can also read CreditCards, you won't get much information but it should be able to read them

dyay108, 2022-11-14:
> 

this is the third reader, and i already soldered the previous 2 and they didn't work. Ended up damaging those boards trying to desolder.

mariomare22, 2022-11-16:
Hello @adonno i have the tag reader from your website and also have orange light and no feed back on readings.
When powering the  blue led flashes briefly but that's it.

Any idea?

adonno, 2022-11-17:
Ok that's wierd. 
It should connect to HA does it connect to it?

Does the card appear in the logs when scanned ? 
Scan a card while logging just to make sure the reader is working properly 
> > 
> 
> 
> 
> this is the third reader and third module, and i already soldered the previous 2 and they didn't work. Ended up damaging those boards trying to desolder.

dyay108, 2022-11-17:
> Ok that's wierd. It should connect to HA does it connect to it?
>
> Does the card appear in the logs when scanned ? Scan a card while logging just to make sure the reader is working properly

No output in the log window when i present a tag to the reader

adonno, 2022-11-19:
and you have been using the cards that are provided with the boards ? 
Or did these come without a card ?

dyay108, 2022-11-19:
Yes, been using the cards that came with the boards and other cards i bought elsewhere.

dyay108, 2022-11-22:
> i don't get the issue and you did use the code that is provided on github without altering it?

correct. the only thing i changed is the ssid and password.

just in case, could u please list out the steps?

adonno, 2022-11-23:
Steps : 

1. solder evrything togehter as per : (in you case ignore the LED and Buzzer) https://github.com/adonno/tagreader/blob/master/Schematics/tag_reader_schematics_v3.2.png 
2. flip the N1 switch to ON on the PN532
3. Connect to a pc and go to : https://adonno.github.io/tagreader 
4. once flashed post the logs

warrap1989, 2022-11-28:
@adonno I followed the steps in your last post but I'm continuing to have the same issue as OP. I wrote out the steps I followed and configuration details below.

On my router, I've set the IP address of the D1 mini to a static address (as part of troubleshooting). HA version is 2022.11.4 and is setup for external access.

Before doing any of this, I removed added devices in integration from previous attempts, uninstalled the ESP Home add-on, deleted the esphome folder under configs and then proceeded with the following steps. 

1. Run the install from https://adonno.github.io/tagreader/
2. After install, assigned network credentials when prompted in https://adonno.github.io/tagreader/
3. Go to Home Assistant and adopt in through ESPHome add-on in Home Assistant
4. After clicking adopt, I was prompted in ESP Home to input network credentials, I used the same ones as step 2
5. Proceed with updates installation in ESP Home add-on, kept log window open and proceeded with next steps
6. After install was complete, added the newly discovered ESP device to my integrations, using the IP in the log file when prompted for the host. integration was added successfully.
7. Used the Clean Tag and Write Tag entities to attempt to write NFC tag that came with the PN532 and downloaded the log files which are attached.

I've tried a number of approaches and I can't seem to get it working. Any insight would be appreciated; it's a really cool project and I'd love to get it up and running :)

[logs_tagreader-0c5c6c install log.txt](https://github.com/adonno/tagreader/files/10106443/logs_tagreader-0c5c6c.install.log.txt)

grambas, 2022-12-30:
Same for me. Everything looks fine. Buzzer reacts to Home Assistant actions like Cancel Writinh, Clean Tag, Write Tag Random, but the reader not reacts to any rfid device
[image]

grambas, 2022-12-30:
> Same for me. Everything looks fine. Buzzer reacts to Home Assistant actions like Cancel Writinh, Clean Tag, Write Tag Random, but the reader not reacts to any rfid device [image]

My Problem was wrong wiring. I needed to change SDA to D2 and CSL to D1

feutl, 2023-01-19:
Got my D1 Minis and Tag Reader today.
Seems not to work, even with flipped D1 and D2 connection. The Tagreade gets power, the buzzer and LEDs work, but no interaction happens when placing a card over the reader after I installed the D1 mini via the webui and integrated it to my HA instance.

thomastalving, 2023-01-24:
I'm having the exact same issue. switching D2 and D1 does not solve it. 

Good to note is that i skipped the buzzer and the led and added a battery module on top of my esp8266, but this device should not interfere with everything.

[Battery Module](https://www.kjell.com/se/produkter/el-verktyg/arduino/stromforsorjning/luxorparts-batterimodul-till-d1-mini-utvecklingskort-p87296?gclid=CjwKCAiAoL6eBhA3EiwAXDom5gvJk2ba6dF3Gt0cyvJ3svdj7NEM5yAxN91hSXCORp63NXQZxrLkkhoCbFUQAvD_BwE&gclsrc=aw.ds)

abdullahvenom21, 2024-07-12:
i have the same issue as above the buzzer reacts to cancel tag,clean tag but the tag reads nothing
[image]
[image]