adonno/tagreader #205: Can't make it work ...

Elmout33, 2022-11-18:
Hi, I follow all steps describe but nothing works ...

I have HA on a Raspberry pi 4 so I can't install from it to the D1 Mini so I try to do so from my computer. It seems ok according to the log in the windows console : 
`INFO Reading configuration C:\Users\lerou\Desktop\tagreader-1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tagreader (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (3.2.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- Wire @ 1.0
|-- Improv @ 1.2.3
|-- NeoPixelBus @ 2.6.9
|   |-- SPI @ 1.0
Building in release mode
Compiling .pioenvs\tagreader\src\main.cpp.o
Linking .pioenvs\tagreader\firmware.elf
Retrieving maximum program size .pioenvs\tagreader\firmware.elf
Checking size .pioenvs\tagreader\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  43.5% (used 35600 bytes from 81920 bytes)
Flash: [=====     ]  48.7% (used 508749 bytes from 1044464 bytes)
Building .pioenvs\tagreader\firmware.bin
Creating BIN file ".pioenvs\tagreader\firmware.bin" using "C:\Users\lerou\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pioenvs\tagreader\firmware.elf"
esp8266_copy_factory_bin([".pioenvs\tagreader\firmware.bin"], [".pioenvs\tagreader\firmware.elf"])
============================================= [SUCCESS] Took 13.68 seconds =============================================INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] COM3 (USB-SERIAL CH340 (COM3))
  [2] Over The Air (tagreader.local)
(number): 1
esptool.py v3.3.1
Serial port COM3
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c8:c9:a3:0a:cd:f7
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x0007dfff...
Compressed 512896 bytes to 348106...
Wrote 512896 bytes (348106 compressed) at 0x00000000 in 8.5 seconds (effective 482.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
[... 275 lines cut ...]
[21:10:37][C][captive_portal:088]: Captive Portal:
[21:10:37][C][mdns:103]: mDNS:
[21:10:37][C][mdns:104]:   Hostname: tagreader-0acdf7
[21:10:37][C][ota:093]: Over-The-Air Updates:
[21:10:37][C][ota:094]:   Address: tagreader.local:8266
[21:10:37][C][api:138]: API Server:
[21:10:37][C][api:139]:   Address: tagreader.local:6053
[21:10:37][C][api:143]:   Using noise encryption: NO
[21:10:37][C][improv_serial:032]: Improv Serial:
[21:10:37][C][wifi_info:009]: WifiInfo IPAddress 'TagReader IP Address'
[21:10:37][C][wifi_info:009]:   Icon: 'mdi:wifi'
[21:10:37][C][wifi_info:011]: WifiInfo SSID 'TagReader Connected SSID'
[21:10:37][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[21:10:57][W][api.connection:083]: ESPHome Logs 2022.11.1 (192.168.1.30): Connection closed
[21:10:57][D][binary_sensor:036]: 'TagReader Status': Sending state OFF`

So I go to HA and a new card was discovered "tagreader-0acdf7". I click on "Adopt" to create a new card with a friendly name. I go to Setting -> Integration -> Device -> Add ESPHOME -> set @IP and have a device added.

But when I pass a nfc card on the nfc reader, nothing appear in Setting -> Tags or in the log on the device ...

I am doing something wrong ? What I have missed ?

Thanks a lot for your help.

But since I got nothing when I pass a card on NFC reader ...

adonno, 2022-11-18:
Your issue is here, [image]

Can you upload picture from your soldering?

Elmout33, 2022-11-18:
Here pictures. 

[image]
[image]
[image]

Elmout33, 2022-11-19:
Hi @adonno, I make the real solidering (yesterday just for test I just plug the component whitout soldering).

My error message disappear :) and my tag are now detected in Home Assistant !
[image]

Thanks a lot for your quick answer !

closed 2022-11-19 by Elmout33