adonno/tagreader #211: Install hangs when writing config to D1 Mini

bsamarketing, 2022-12-04:
I am trying to install thins through ESP home in Home assistant. I past the config and add my wifi details.

On hitting install, it start to install but always hangs halfway through (Though not always at the same place)

Log file for the install is below:

INFO Reading configuration /config/esphome/tagreader.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tagreader (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Removing Improv @ 1.2.1
INFO Removing Improv @ 1.2.1
Library Manager: Improv@1.2.1 has been removed!
INFO Improv@1.2.1 has been removed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- Wire @ 1.0
|-- Improv @ 1.2.3
|-- NeoPixelBus @ 2.6.9
|   |-- SPI @ 1.0
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/api_connection.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/api_server.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/list_entities.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/proto.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/api/user_services.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/button/button.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/dashboard_import/dashboard_import.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/esp8266/core.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/esp8266/gpio.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/esp8266/preferences.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/esp8266_pwm/esp8266_pwm.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/i2c/i2c.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/improv_serial/improv_serial_component.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/addressable_light.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/automation.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/esp_color_correction.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/esp_hsv_color.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/esp_range_view.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/light_call.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/light_json_schema.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/light_output.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/light/light_state.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/components/logger/logger.cpp.o
[... 50 lines cut ...]
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/core/helpers.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/core/log.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/core/scheduler.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/esphome/core/util.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/src/main.cpp.o
Generating LD script /data/tagreader/.pioenvs/tagreader/ld/local.eagle.app.v6.common.ld
Compiling /data/tagreader/.pioenvs/tagreader/libe4d/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/libe4d/ESPAsyncTCP-esphome/ESPAsyncTCP.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/libe4d/ESPAsyncTCP-esphome/ESPAsyncTCPbuffer.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/libe4d/ESPAsyncTCP-esphome/SyncClient.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/libe4d/ESPAsyncTCP-esphome/tcp_axtls.c.o
Compiling /data/tagreader/.pioenvs/tagreader/libaf0/Hash/Hash.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/BearSSLHelpers.cpp.o
Archiving /data/tagreader/.pioenvs/tagreader/libaf0/libHash.a
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/CertStoreBearSSL.cpp.o
Archiving /data/tagreader/.pioenvs/tagreader/libe4d/libESPAsyncTCP-esphome.a
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
Compiling /data/tagreader/.pioenvs/tagreader/lib67b/ESP8266WiFi/ESP8266WiFiMulti.cpp.o

This is where it hangs

Any suggestions?

DRnooberson, 2022-12-06:
Hi, 
I assume you are running HA on a raspberry pi, I've had similar issues previously and it was solved by increasing the swap memory size. You can read through this post for more details: https://community.home-assistant.io/t/how-to-increase-the-swap-file-size-on-home-assistant-os/272226

bsamarketing, 2022-12-07:
I sorted this. I didn't realise I needed to install the ESP Home integration in HA as well as the add on Noob error! Doh!

closed 2022-12-10 by adonno