adonno/tagreader #214: Timed out waiting for readiness from PN532 error

DRnooberson, 2022-12-06:
Hello,

Lately I've been trying to install the provided code on a D1 mini with a PN532 board. In the past, I have been successful on breadboard but now that I want to use a more permanent package (using the 3d printed case) I am unable to flash it again.

Everytime I try to install the code I end up in an infinite loop showing this message:
[image]

**Things I have tried:**
- Change D1 mini for another manufacturer
- Change PN532 for another manufacturer
- Re-do the wiring multiple time
- Tried flashing using all 4 methods (Wirelessly, Computer running HA, Through the browser ESP Web tool, Manual install using ESPtool) all with same results.

**Notes:** 
In the installation process, I can see that the D1 mini connects properly to the Wifi network so I don't think the problem would come from that part.
Since I has work in the past, could it be due to HA or ESPhome version being different? I am running the following versions:
- Home Assistant Core 2022.11.4
- ESPHome 2022.11.4

Any help would be greatly appreciated! Thank you :)

adonno, 2022-12-10:
may i Suggest you remove the LED and the buzzer and only try with the PN532 ?

DRnooberson, 2022-12-13:
> may i Suggest you remove the LED and the buzzer and only try with the PN532 ?

Didn't seem to have change anything sadly... Could it be due to the ESP version I am running? Have you had any chance flashing a board recently?

unios0932, 2023-05-07:
@[DRnooberson](https://github.com/DRnooberson), have you solved this issue?
It looks I have the same issue.

I am trying to do a first start on ESPhome on HA but on the last part the API can't connect to HA.
The reason I see in verbose mode probably indicate problems with PN532 module
I have an issue with "Component pn532 took a long time for an operation".
On the other issue on ESPhome issues - [#3281 PN532 over I2C stops responding after a while](https://github.com/esphome/issues/issues/3281) @mrkylegp said that "it is there is a problem with the module on a hardware/firmware level". Does that mean that the this specific PN532 hardware I have is faulty and I need to try with another version of this module or I need to do something else? I double checked but might be the problem with soldering?

### Which version of ESPHome has the issue?
2023.4.4
### What type of installation are you using?
On Raspberry Pi 4
### Which version of Home Assistant has the issue?
Home Assistant 2023.5.2 
Supervisor 2023.04.1 
Operating System 10.1

Here is the log in a verbose mode:

```
INFO Successfully compiled program.
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:05:7b:81
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00094fff...
Compressed 607712 bytes to 415159...
Wrote 607712 bytes (415159 compressed) at 0x00000000 in 9.8 seconds (effective 497.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[21:36:41]\x90nn$`nr\x8e\x92\x92nrr\x92`p\xf2n\x82r\x8cbbn\xc2nb\xc4\xf2nn'l`\x90nn$`nr\x8e\x92\x92nrl`r\x92\x92nrl`\x9cl$$l`\xfc\x82n\x92`
[21:36:41]SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[21:36:41][I][logger:259]: Log initialized
[21:36:41][C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[21:36:41][I][app:029]: Running through setup()...
[21:36:41][V][app:030]: Sorting components by setup priority...
[21:36:41][I][i2c.arduino:183]: Performing I2C bus recovery
[21:36:41][D][template.switch:049]:   Restored state OFF
[21:36:41][D][switch:016]: 'TagReader Buzzer Enabled' Turning OFF.
[21:36:41][D][switch:055]: 'TagReader Buzzer Enabled': Sending state OFF
[21:36:41][D][template.switch:049]:   Restored state OFF
[21:36:41][D][switch:016]: 'TagReader LED enabled' Turning OFF.
[21:36:41][D][switch:055]: 'TagReader LED enabled': Sending state OFF
[21:36:41][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[21:36:41][C][light:035]: Setting up light 'TagReader LED'...
[21:36:41][D][light:035]: 'TagReader LED' Setting:
[21:36:41][D][light:040]:   Color mode: RGB
[21:36:41][D][light:084]:   Transition length: 1.0s
[21:36:41][C][pn532:018]: Setting up PN532...
[21:36:41][V][pn532:249]: Reading ACK...
[21:36:41][V][pn532:260]: ACK valid: YES
[21:36:41][V][pn532_i2c:043]: Reading response
[21:36:41][V][pn532:265]: Sending NACK for retransmit
[21:36:41][V][pn532_i2c:049]: Reading response of length 5
[21:36:41][D][pn532:036]: Found chip PN532
[21:36:41][D][pn532:037]: Firmware ver. 1.6
[... 233 lines cut ...]
[21:36:52][C][wifi_info:009]:   Icon: 'mdi:wifi'
[21:36:52][V][wifi_info:009]:   Unique ID: 'cc50e3057b81-wifiinfo-ip'
[21:36:52][C][wifi_info:011]: WifiInfo SSID 'TagReader Connected SSID'
[21:36:52][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[21:36:52][V][wifi_info:011]:   Unique ID: 'cc50e3057b81-wifiinfo-ssid'
[21:36:52][V][pn532:249]: Reading ACK...
[21:36:52][V][pn532:260]: ACK valid: YES
[21:36:52][V][pn532_i2c:043]: Reading response
[21:36:52][V][pn532_i2c:032]: Timed out waiting for readiness from PN532!
[21:36:52][V][pn532:271]: Turning RF field OFF
[21:36:52][V][pn532:249]: Reading ACK...
[21:36:52][V][pn532:260]: ACK valid: YES
**[21:36:52][V][component:204]: Component pn532 took a long time for an operation (0.11 s).**
[21:36:52][V][component:205]: Components should block for at most 20-30ms.
[21:36:53][V][pn532:249]: Reading ACK...
[21:36:53][V][pn532:260]: ACK valid: YES
[21:36:53][V][pn532_i2c:043]: Reading response
[21:36:53][V][pn532_i2c:032]: Timed out waiting for readiness from PN532!
[21:36:53][V][pn532:271]: Turning RF field OFF
[21:36:53][V][pn532:249]: Reading ACK...
[21:36:53][V][pn532:260]: ACK valid: YES
...
```

In case on need for more logs I am ready to send here.

adonno, 2023-05-07:
would you mind having the logs less verbose and trying to scan a card ?  
somtimes depdning on your HA configuration you wil have to add the tagreader manually 

Settings --> devices & Services --> Add integration --> Esphome --> enter the ipaddress

unios0932, 2023-05-07:
Finally it works! Last time I did that it didn't show all parameters of device but now it looks good.
On the other hand tagreader in esphome is still offline. Apart of that tagreader in HA still works. Thank you!