adonno/tagreader #215: Requesting tag read failed! error

bsamarketing, 2022-12-06:
Hi

The code is now installed, but I am getting a Requesting tag read failed! error and it wont read tags.

I have checked the wiring, and all looks goo, (SCA - D2 SDL D1 + 5V and Gnd) and the tage reader is def powered up.

Any suggestions

Logs below.

INFO Reading configuration /config/esphome/tagreader.yaml...
INFO Starting log output from 192.168.123.22 using esphome API
INFO Successfully connected to 192.168.123.22
[16:12:01][I][app:102]: ESPHome version 2022.11.4 compiled on Dec  6 2022, 11:06:21
[16:12:01][I][app:104]: Project adonno.tag_reader version 1.4
[16:12:01][C][wifi:504]: WiFi:
[16:12:01][C][wifi:362]:   Local MAC: EC:FA:BC:59:48:27
[16:12:01][C][wifi:363]:   SSID: [redacted]
[16:12:01][C][wifi:364]:   IP Address: 192.168.123.22
[16:12:01][C][wifi:365]:   BSSID: [redacted]
[16:12:01][C][wifi:367]:   Hostname: 'tagreader-594827'
[16:12:01][C][wifi:369]:   Signal strength: -58 dB ▂▄▆█
[16:12:01][C][wifi:373]:   Channel: 10
[16:12:01][C][wifi:374]:   Subnet: 255.255.255.0
[16:12:01][C][wifi:375]:   Gateway: 192.168.123.146
[16:12:01][C][wifi:376]:   DNS1: 0.0.0.0
[16:12:01][C][wifi:377]:   DNS2: 0.0.0.0
[16:12:04][W][pn532:090]: Requesting tag read failed!
[16:12:04][C][logger:293]: Logger:
[16:12:04][C][logger:294]:   Level: DEBUG
[16:12:04][C][logger:295]:   Log Baud Rate: 115200
[16:12:04][C][logger:296]:   Hardware UART: UART0
[16:12:07][W][pn532:090]: Requesting tag read failed!
[16:12:07][D][rtttl:094]: Playback finished
[16:12:07][C][i2c.arduino:052]: I2C Bus:
[16:12:07][C][i2c.arduino:053]:   SDA Pin: GPIO4
[16:12:07][C][i2c.arduino:054]:   SCL Pin: GPIO5
[16:12:07][C][i2c.arduino:055]:   Frequency: 400000 Hz
[16:12:07][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[16:12:10][W][pn532:090]: Requesting tag read failed!
[16:12:13][W][pn532:090]: Requesting tag read failed!
[16:12:13][C][template.switch:058]: Template Switch 'TagReader Buzzer Enabled'
[16:12:13][C][template.switch:058]:   Icon: 'mdi:volume-high'
[16:12:13][C][template.switch:059]:   Restore State: YES
[16:12:13][C][template.switch:060]:   Optimistic: YES
[16:12:16][W][pn532:090]: Requesting tag read failed!
[16:12:16][C][template.switch:058]: Template Switch 'TagReader LED enabled'
[16:12:16][C][template.switch:058]:   Icon: 'mdi:alarm-light-outline'
[16:12:16][C][template.switch:059]:   Restore State: YES
[16:12:16][C][template.switch:060]:   Optimistic: YES
[16:12:19][W][pn532:090]: Requesting tag read failed!
[16:12:19][C][esp8266_pwm:022]: ESP8266 PWM:
[16:12:19][C][esp8266_pwm:023]:   Pin: GPIO13
[16:12:19][C][esp8266_pwm:024]:   Frequency: 988.0 Hz
[16:12:22][W][pn532:090]: Requesting tag read failed!
[16:12:25][W][pn532:090]: Requesting tag read failed!
[16:12:28][W][pn532:090]: Requesting tag read failed!
[16:12:28][C][light:104]: Light 'TagReader LED'
[16:12:28][C][light:106]:   Default Transition Length: 1.0s
[16:12:28][C][light:107]:   Gamma Correct: 2.80
[16:12:31][W][pn532:090]: Requesting tag read failed!
[16:12:34][W][pn532:090]: Requesting tag read failed!
[16:12:34][C][restart.button:017]: Restart Button 'TagReader Restart'
[16:12:37][W][pn532:090]: Requesting tag read failed!
[16:12:37][C][pn532:347]: PN532:
[16:12:37][C][pn532:359]:   Update Interval: 1.0s
[16:12:37][C][pn532_i2c:128]:   Address: 0x24
[16:12:40][W][pn532:090]: Requesting tag read failed!
[16:12:40][C][status:034]: Status Binary Sensor 'TagReader Status'
[16:12:40][C][status:034]:   Device Class: 'connectivity'
[16:12:43][W][pn532:090]: Requesting tag read failed!
[16:12:43][C][version.text_sensor:021]: Version Text Sensor 'TagReader ESPHome Version'
[16:12:43][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[16:12:46][W][pn532:090]: Requesting tag read failed!
[16:12:46][C][rtttl:018]: Rtttl
[16:12:49][W][pn532:090]: Requesting tag read failed!
[16:12:52][W][pn532:090]: Requesting tag read failed!
[16:12:55][W][pn532:090]: Requesting tag read failed!
[16:12:58][W][pn532:090]: Requesting tag read failed!

bsamarketing, 2022-12-06:
Scratch That , not sure what I did but its working now!

adonno, 2022-12-10:
hm looks like it was working on the startup and now it sin't . that could be a faulty connection between PN532 and D1 mini 
can you please look ad your soldering 

Like i said, it worked on boot and them something changed (you moved it?) and now its not detecting PN532 anymore