adonno/tagreader #217: Upgrading to esp32

flouck, 2022-12-12:
Hi!
Do you think it is possible to buy a small esp32 and dismount your tagreader to replace the esp8266?
My goal is to have a bluetooth proxy but I don't want to manage another device plugged into a power plug socket with a usb adapter. It's too much in my home now! 
So, I am trying to rethink of all my devices which are the ones that could do two things at once. 
What do you think?

gmbuell, 2022-12-26:
FWIW I just did a build using a Lolin C3 Mini since it shares the same form factor as the D1. Only changes needed to the yaml were specifying the board and the LED pin (since the C3 Mini has a built in LED).

flouck, 2022-12-26:
woow! And it worked? So it does all the tagreader functionalities and besides oh that, you use it as a Bluetooth proxy?

lunarok, 2023-01-04:
will be interested by this also

gmbuell, 2023-01-04:
I don't personally use the Bluetooth proxy functionality but everything else works. I expect enabling bluetooth proxy would be as simple as adding those lines to the yaml config

flouck, 2023-01-05:
May I know how do you use the Bluetooth then? Out of curiosity as I am always looking for new use cases 😉

gmbuell, 2023-01-05:
I don't use bluetooth.

As another note: the buzzer doesn't work right on ESP32-C3 boards on the current release of ESPHome. You need a patch which has been merged upstream but not released yet. You can use it with:
external_components:
  - source: github://pr#3820
    components: [ ledc ]

gmbuell, 2023-01-05:
No technical reason. I had some C3-minis laying around.

flouck, 2023-06-24:
Hello @gmbuell,
Could you please give me the pinout comparison between the esp8266 D1 mini and the esp32 Cr mini?
Not sure how to replace my esp8266 by an esp32.
Thank you!!!!!!!

leinich, 2023-12-23:
I documented my Pin Layout and adjustments to yaml here: 
https://github.com/leinich/tagreader
@flouck