adonno/tagreader #218: 1 reader is running automations for all readers

LuxStatera, 2022-12-25:
I currently use 3 different tags with 6 different readers. I have them setup so that if tag1 is scanned by reader1 then it runs automation1. Scanning tag1 on reader2 will run automation2 and so on. 

Now when I scan tag1 on reader1 it runs automation1 through automation6. How can I fix this?

LuxStatera, 2023-08-18:
It is happening again. Unfortunately I don't remember what fixed it last time. Maybe restarting HA. I tried that but I'm still having the same issue

tim-vk, 2024-07-30:
Old topic but: isn't this a home assistant question? (my suggestion: close this issue & ask at the home assistant forums).

adonno, 2024-07-30:
Male sire you check if you used the deviced id in yojr automations and verify your traces

closed 2024-07-30 by LuxStatera