adonno/tagreader #222: Wifi Disconnecting

conorokelly, 2023-01-09:
I am getting very regular, but inconsistent, disconnects from WIFI with a brand new reader. The disconnects are caused by scanning a tag. 
Sometimes the action is received by home assistant - sometimes not, sometimes it disconnects (the majority of the time). It is quite slow when it does work - long pauses - compared to swiping a tag on the home assistant app.
I have tried different cables, different power etc. Tried changing physical placement etc. 
The particular tags can work sometimes and not others.

conorokelly, 2023-01-10:
This is a log from esphome along with a disconnect after scan:

NFO Reading configuration /config/esphome/tagreader-8bd140.yaml...
INFO Starting log output from tagreader-8bd140.local using esphome API
WARNING Can't connect to ESPHome API for tagreader-8bd140.local: Timeout while connecting to ('192.168.178.114', 6053)
INFO Trying to reconnect to tagreader-8bd140.local in the background
INFO Successfully connected to tagreader-8bd140.local
[17:09:47][I][app:102]: ESPHome version 2022.12.3 compiled on Jan 10 2023, 17:02:33
[17:09:48][I][app:104]: Project adonno.tag_reader version 1.4
[17:09:48][C][wifi:504]: WiFi:
[17:09:48][C][wifi:362]:   Local MAC: 40:22:D8:8B:D1:40
[17:09:48][C][wifi:363]:   SSID: '2.4 Nolite te bastardes'[redacted]
[17:09:48][C][wifi:364]:   IP Address: 192.168.178.114
[17:09:48][C][wifi:365]:   BSSID: DC:15:C8:23:3F:52[redacted]
[17:09:48][C][wifi:367]:   Hostname: 'tagreader-8bd140'
[17:09:48][C][wifi:369]:   Signal strength: -71 dB ▂▄▆█
[17:09:48][C][wifi:373]:   Channel: 1
[17:09:48][C][wifi:374]:   Subnet: 255.255.255.0
[17:09:48][C][wifi:375]:   Gateway: 192.168.178.1
[17:09:48][C][wifi:376]:   DNS1: 192.168.178.1
[17:09:48][C][wifi:377]:   DNS2: 0.0.0.0
[17:09:48][C][logger:293]: Logger:
[17:09:48][C][logger:294]:   Level: DEBUG
[17:09:48][C][logger:295]:   Log Baud Rate: 115200
[17:09:48][C][logger:296]:   Hardware UART: UART0
[17:09:48][C][i2c.arduino:052]: I2C Bus:
[17:09:48][C][i2c.arduino:053]:   SDA Pin: GPIO4
[17:09:48][C][i2c.arduino:054]:   SCL Pin: GPIO5
[17:09:48][C][i2c.arduino:055]:   Frequency: 400000 Hz
[17:09:48][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[17:09:48][C][template.switch:076]: Template Switch 'TagReader Buzzer Enabled'
[17:09:48][C][template.switch:078]:   Icon: 'mdi:volume-high'
[17:09:48][C][template.switch:098]:   Restore Mode: restore defaults to OFF
[17:09:48][C][template.switch:059]:   Restore State: YES
[17:09:48][C][template.switch:060]:   Optimistic: YES
[17:09:48][C][template.switch:076]: Template Switch 'TagReader LED enabled'
[17:09:48][C][template.switch:078]:   Icon: 'mdi:alarm-light-outline'
[17:09:48][C][template.switch:098]:   Restore Mode: restore defaults to OFF
[17:09:48][C][template.switch:059]:   Restore State: YES
[17:09:48][C][template.switch:060]:   Optimistic: YES
[17:09:48][C][esp8266_pwm:022]: ESP8266 PWM:
[17:09:48][C][esp8266_pwm:023]:   Pin: GPIO13
[17:09:48][C][esp8266_pwm:024]:   Frequency: 988.0 Hz
[17:09:48][C][light:104]: Light 'TagReader LED'
[17:09:48][C][light:106]:   Default Transition Length: 1.0s
[17:09:48][C][light:107]:   Gamma Correct: 2.80
[17:09:48][C][restart.button:017]: Restart Button 'TagReader Restart'
[17:09:48][C][pn532:347]: PN532:
[17:09:48][C][pn532:359]:   Update Interval: 1.0s
[17:09:48][C][pn532_i2c:128]:   Address: 0x24
[17:09:48][C][status:034]: Status Binary Sensor 'TagReader Status'
[17:09:48][C][status:034]:   Device Class: 'connectivity'
[17:09:48][C][version.text_sensor:021]: Version Text Sensor 'TagReader ESPHome Version'
[17:09:48][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[17:09:48][C][rtttl:018]: Rtttl
[17:09:48][C][captive_portal:088]: Captive Portal:
[17:09:48][C][mdns:103]: mDNS:
[17:09:48][C][mdns:104]:   Hostname: tagreader-8bd140
[17:09:48][C][ota:093]: Over-The-Air Updates:
[17:09:48][C][ota:094]:   Address: tagreader-8bd140.local:8266
[17:09:48][C][api:138]: API Server:
[17:09:48][C][api:139]:   Address: tagreader-8bd140.local:6053
[17:09:48][C][api:143]:   Using noise encryption: NO
[17:09:48][C][improv_serial:032]: Improv Serial:
[17:09:48][C][wifi_info:009]: WifiInfo IPAddress 'TagReader IP Address'
[17:09:48][C][wifi_info:009]:   Icon: 'mdi:wifi'
[17:09:48][C][wifi_info:011]: WifiInfo SSID 'TagReader Connected SSID'
[17:09:48][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[17:11:04][D][pn532:283]: Mifare ultralight
[17:11:05][D][light:035]: 'TagReader LED' Setting:
[17:11:05][D][light:040]:   Color mode: RGB
[17:11:05][D][light:046]:   State: ON
[17:11:05][D][light:050]:   Brightness: 100%
[17:11:05][D][light:057]:   Red: 0%, Green: 100%, Blue: 0%
[17:11:05][D][light:077]:   Flash length: 0.5s
[17:11:05][D][pn532:162]: Found new tag '04-D2-4D-20-B6-2A-81'
[17:11:05][D][pn532:166]:   NDEF formatted records:
[17:11:05][D][pn532:168]:     U - https://open.spotify.com/track/0DiWol3AO6WpXZgp0goxAV?si=9893eb6000c94598
[17:11:05][D][pn532:168]:     T - Daft Punk discovery 
[17:11:05][D][pn532:295]: Waiting to read next tag
INFO tagreader-8bd140.local: Ping timed out!
INFO Disconnected from ESPHome API for tagreader-8bd140.local
WARNING Disconnected from API
INFO Successfully connected to tagreader-8bd140.local
[17:11:44][D][api:102]: Accepted 192.168.178.69

conorokelly, 2023-01-11:
I've spent a good deal of my spare time for the last week troubleshooting this. Emailed Andrea directly etc. I think I need to return the device unless someone has some advice.

nielsrune, 2023-02-07:
In the beginning I had issued getting the chip to even get an IP address and connect to Home Assistant.

I would suggest you to hard code a static (local) IP address etc. and see if it helps. Both my tagreaders are working as expected since I did this.

This, however, will require you to edit the yaml and flash it onto the chip, but it is not that complicated:
```
manual_ip:
  static_ip: 192.168.xx.yy
  gateway: 192.168.xx.yy
  subnet: 255.255.255.0
  dns1: 1.1.1.3 <-- Cloudflare's dns. Change to your linking
  dns2: 1.0.0.3 <-- same as above
```

It it works, I guess the issue is out of Adonno's hands.

foothook, 2023-11-21:
Hi i stumbled on similar problems too (November 2023) with 12 NFC readers directly bought from adonno shop).

Newest available homeassistant version and ESPHome version. Tried with original firmware (as adonno has shipped them) and adopting them in ESPHome for the newest update as well

The readers work normaly when you place and remove an tag immediately.

In my use case the tags are placed, left there for some time and than removed and a new tag is placed.
The expected behaviour is that i can trigger automatitions from the 3 events
-TAG A read
-TAG removed
-TAG B read

This works when the actions are executed in a time span of a few seconds.

But the NCF reader locks up when the tag is left there for multiple seconds or minutes.

It becomes unresponsive at PING.
When the tag is removed, the reader recovers from this lock up but only after several seconds, somethimes it even reboots (reconizable from the reboot beep).

This is unacceptable for my usecase because we need a reliable recognition of the TAG read and TAG removed action.
While locked (the few seconds to the 1,5minutes in case of reboot) they don`t read so my usecases dont work.

I have tried a lot of troubleshooting by myself, change powersource and cable, checking tags and different readers, trying to figure out if i can fix something on YAML.
Tried do reach out to andrea adonno via email for support.

It seems to be the same problem as the OP, opened on January, but in my use case even with worse effects on the use case.
My guess is that the NFC reader continously reads the Tag and locks up because its reading quicker than the ESP can process and so it causes to lock up.
I would try to limit the read rate to something processable, but i am lacking the skills do dig so deeply into the code myself.

Maybe there is someone that could help me with this?

only alternative is request a refund for the adonno readers and search some other nfc solution, like arduino...

Nasimovy, 2023-11-21:
according to esphome 
"update_interval (Optional, [Time](https://esphome.io/guides/configuration-types#config-time)): The duration of each scan on the PN532. This affects the duration that the individual binary sensors stay active when they’re found. If a device is not found within this time window, it will be marked as not present. Defaults to 1s."

so you could try and change the value to more than 1sec to try but this shouldn't be the problem...

how do you do this?
under this line of code:
pn532_i2c:

add this
  update_interval: 1s

conorokelly, 2023-11-21:
Op here.
Like you these issues were caused by leaving the tag on the reader.
The reader, like yours, stopped responding to pings with the card left in place. I surmised (perhaps incorrectly) that the card was blocking signal and the reader design does not accommodate leaving the card in place. I adapted my use (jukebox) by placing the reader vertically so that the card is simply scanned and removed.

foothook, 2023-11-24:
> Op here. Like you these issues were caused by leaving the tag on the reader. The reader, like yours, stopped responding to pings with the card left in place. I surmised (perhaps incorrectly) that [...]

Hi @conorokelly , thanks for reply.
I wish to update you (and other readers of this issue) about my findings. I did some yaml code simplifications with the help of @Nasimovy and kvi on the esphome discord support channel.

We have removed all functionallity (buzzer led etc except the simple on_tag and tag_removed functionallity). Then one of my readers works whitout locking up when a tag is left on (and it became very very responsive, reads tags very quickly)

But i have 12 readers from another batch that keep locking up, so it seems that they could be from a defective batch,
I reached out to andrea adonno for support via email and hope to hear back soon

marioschulz93, 2024-11-01:
@foothook I'm currently facing the same problem. Would you be so kind as to upload the customized yaml code?

Nasimovy, 2024-11-01:
@marioschulz93 

The issue that was present with the modules are that the tag reader is too close to the antenna of the ESP, to solve the issue you have to separate the tag reader away from the motherboard

marioschulz93, 2024-11-01:
@Nasimovy 

thanks for the quick reply. 

However, that makes the solution difficult, I bought the sandwich version, here the circuit boards are firmly soldered. 

Do you see any way to shield the boards from each other?

Nasimovy, 2024-11-01:
@marioschulz93 

In the ESPhome discord there was a user who successfully desoldered the tag reader from the esp board, and he reported that the original firmware was working on the modified tag reader.

I still can understand that it's not an easy task, you could try contacting @adonno

foothook, 2024-11-01:
> @foothook I'm currently facing the same problem. Would you be so kind as to upload the customized yaml code?

Hi,
the custom firmware did only help for troubleshooting but did not resolve it completely, so i did switch back to original after solving it by spacing out.

as mentioned by @marioschulz93 the solution was to dissolder the nfc and space it away from the ESP because it seems that on the sandwich model the NFC module is interfering the wifi connection in the moment a tag is on the reader.

Spacing out was the only fix for me, dissoldering is a bit tricky on the prefactored board without the right tools as you could overheat and destroy the conductor tracks or components.

maybe reach out to @adonno for help with the sandwich board.

cmprmsd, 2025-05-22:
@adonno 
I'm also experiencing this problem. As desoldering isn't an option due to the lack of proper case printing hardware I'd like to ask if there are any software workarounds for this. 

Is it possible to increase the WiFi signal power or reduce the nfc power? In this thread update intervals were mentioned. Does this solve the issues for the other people in this thread? 

I built a tonibox with nfc chips and this nfc reader and it keeps doing the sound in the middle of the night, which wakes my kid. 😳

JohnMcLear, 2025-07-03:
[image]

I'm experiencing reboots every 1.5 hrs w/ my device...  

To get information I include debug: with some text_Sensors..

The pn532 code is causing a crash on this board and other PN532 board...

The high loop time happens prior to the reboot..  The ESP is saving itself by rebooting..

If you comment out the pn532 code the device is stable.

[image]

FWIW I tested and confirmed this on a sandwich version but I have access to multiple ESPHome friendly boards and PN532 devices and confirm they behave the same.  I believe this issue is an ESPHome issue or PN532 driver issue (which is still an ESPHome issue).  @adonno please chime in if useful :)

foothook, 2025-07-05:
since I see some new activity on this issue, I confirm that it has to be a physical proximity issue of the sandwich board.

My solution with the dissoldered and spaced out boards works since 2023 without issues. The NFC reader board is places  2cm away from the ESP board with a wire connection.

One Reader i left as original ( because its another use case where it has minor impact)  has still the same problem.

So if you need only single read events the sandwich boards do probably work for you.

But if you have use cases where tag remains placed on the reader and you want to track also the removed action, the bug of the  design error of the sandwich board is impacting and you should opt for the self made board option.

I don't know if @adonno in the meantime has changed the design and has a new board version without the issue, but if not i do not recommend the sandwich board version because of the design flaw.

JohnMcLear, 2025-07-05:
> since I see some new activity on this issue, I confirm that it has to be a physical proximity issue of the sandwich board.
> 
> My solution with the dissoldered and spaced out boards works since 2023 without issues. The NFC reader board is places 2cm away from the ESP board with a wire connection.
> 
> One Reader i left as original ( because its another use case where it has minor impact) has still the same problem.
> 
> So if you need only single read events the sandwich boards do probably work for you.
> 
> But if you have use cases where tag remains placed on the reader and you want to track also the removed action, the design flaw of the sandwich board is impacting and you should opt for the self made board option.
> 
> I don't know if [@adonno](https://github.com/adonno) in the meantime has changed the design and has a new board version without the issue, but if not i do not recommend the sandwich board version because of the design flaw.

I will run the same tests again with the pn532 board moved ~2cm away from the esp32 to confirm :)  ETA test results 4pm Sunday 6th July BST.

JohnMcLear, 2025-07-06:
> I will run the same tests again with the pn532 board moved ~2cm away from the esp32 to confirm.
>
> ETA test results 4pm Sunday 6th July BST.

I'm seeing the same results with the device ~2cm away.  I'm not saying it's not the cause but I just don't think its the only issue.  Another test you can do is get some ferrite sheet material and stick that between your PN532 and ESP.

https://www.amazon.co.uk/Adhesive-Ferrite-Ferrous-Magnetic-Movement as an example.

That way you can easily run your own test and see if your results change :)