adonno/tagreader #223: Missing documentation about Apple Music, Spotify and Sonos

tetienne, 2023-01-13:
Hi, I just discovered in the configuration an [occurence](https://github.com/adonno/tagreader/blob/master/tagreader.yaml#L266) of theses services. 

I was not able to find any explanation about this. What’s the goal of these lines?

m01, 2023-01-14:
I think it's related to these:
* https://github.com/adonno/tagreader/pull/156
* and maybe https://github.com/adonno/tagreader/pull/162/files which introduces https://github.com/adonno/tagreader/blob/master/blueprints/PlayMusicTag.yaml

My best guess is that you can import and setup the Blueprint in Home Assistant, and then call the `esphome.tagreader_xxyyzz__write_music_tag` service with a Spotify/Apple Music/Sonos URL and put a tag on the reader to write it. Then I suspect scanning the tag will result in music playing.

It looks pretty neat.

feutl, 2023-01-19:
I am also struggling a little with no documentation especially with the music addon.
I do understand the tags how they get added and how they trigger automation, but the write random stuff on the tag confuses me.
Could someone at least make a video or write something besides code to explain how this gets used properly and what possibilities it has?
Thanks

baurmatt, 2023-02-12:
I'm interested in this as well! @luka6000 and @adonno seem to be the authors of those code. Could you please add some documentation for it? Thanks a lot!

adonno, 2023-02-12:
i wanted but didn't find the time to yet , ive had eventful months and still will until Q3 /2023 so please don't expect anything before it. maybe i'll get around to it

luka6000, 2023-02-13:
Till official docs your best bet is to check this
https://community.home-assistant.io/t/tagreader-jukebox-old-dog-new-tricks/407855/1

rayan1812109, 2023-04-11:
"If you're tired of manually transferring your music between different platforms, MusConv can save you a ton of time and hassle. Works like a charm: http://www.musconv.com/ #musictransfer #musictool #musconv"