adonno/tagreader #226: Can't read NFC-Tags

BjGoCraft, 2023-02-20:
Hi,
Everything is connected right (I tested every connection with a multimeter). The switch 1 on the reader module is on (up) and switch 2 off:
[image]
[image]
[image]
[image]

The ESP is shown in HomeAssistant as a device of the service ESPHome. So the ESP is recognized. I once connected a LED to it and I could control it with the TagReaderLED switch within Home Assistant, so HomeAssistant can communicate with the ESP. But when I put a NFC-Card on the reader, there is no event in the logbook. The NFC-Card works fine with my smartphone (it's the one the Reader came with).

bbrendon, 2023-03-26:
I was running a very old version of this project and mine was working until today when I updated it because I didn't have the LED and Buzzer enabled. 

For me, the esphome appears in hass but like you, the NFC reader no longer works. I'm not sure what specifically the issue is yet. I'm still trying to work it out.

BjGoCraft, 2023-03-26:
For me it turned out to be bad soldering. Even though I checked every connection with a multimeter, resoldering the NFC-Reader to the cables did the job for me.

closed 2023-03-26 by BjGoCraft

bbrendon, 2023-03-26:
After some closer looking, mine is wired completely different. It looks like a lot has changed since I built mine. So I guess I find old firmware or re-wire it.

chernerd, 2023-11-17:
@BjGoCraft when you re-soldered did you have to remove the jumper pins and directly solder the wires? How did you resolder?