adonno/tagreader #231: Scanning a card from the HA app gives a different ID than from the tagreader

ponchohoncho, 2023-03-20:
I bought some NTAG215 cards for my RFID jukebox setup. I added and wrote a card to HA using the mobile app, and added the GUID-looking ID to my automation but when I scan the card with the tagreader, I get a different ID. Here's a screenshot of me scanning the same card from the app and from the tagreader

https://imgur.com/a/OgDTWvb

I guess that this is because the HA app reads the card's ID and then looks up the mapping to the HA ID for it, where the tagreader only passes the card's ID to HA.

Is this expected behavior? 

What it means is that:
1. If I want to be able to scan the cards with my phone, my automation needs to have two entries per card, one with the card ID and one with the HA ID.
2. The Tags section in HA has double entries. One set of nicely-named entries for when I create and write the card, and another set of hex IDs from when they're scanned with tagreader
3. The only way to get the ID of each card is to listen for the tag_scanned event, scan it with the tagreader, and then note it down. This is cumbersome as my card printer and tagreader jukebox are on opposite ends of the house.

screenagerbe, 2023-11-28:
I ran into the exact same issue, kind of annoying tbh.

Lifth1309, 2024-11-02:
I've got a similar issue. I've set up a lot of tags (Hundreds. Jukebox and a lot of recipes - i track the majority of my macros this way) via tagreader in yaml. I'm looking for a way to scan these with my android-phone as well, but scanning with the phone only results in an "empty tag" reading. I assume thats the same problem as described here. I'm looking for a way to use my phone to scan the existing tags witout having to set them all up again via the ui. Is that possible? I would even double up yaml config in different spots (although thats inconvienient), but ui clicing without copy&paste isn't an option.