adonno/tagreader #234: web.esphome.io says "Failed to execute 'open' on 'SerialPort': Failed to open serial port.

zachboatwright, 2023-03-29:
This is my first time doing any projects with ESPHome and also my first time soldering, so please forgive me if this is a stupid question. I've spent a few days trying to find an answer on my own and am having no luck. I have everything soldered together and am trying to install the YAML using web.esphome.io and keep getting the above error when attempting to connect to the device. I've tried a separate ESP8266 chip with no soldering on it and am getting the same error. I've also tried using separate USB cables and also connecting it directly to the intel NUC that I'm running HA on, and have had issues with that. I can't remember the exact error message but it also had to do with the serial port. Any help would be appreciated.

adonno, 2023-05-07:
Can you send In picture of the soldering ? 
Main cause could be a short (almost impossible to see) 

A good practice would be to unsolder everything and test the ESP, then add the PN532 If both work add component one by one until you find the faulty one

Hintenhaus04, 2023-05-07:
I got it working by plugging it into my machine with is running homeassistaint and from there doing every thing wirelessly. I know it is not the perfect solution but it works

zachboatwright, 2023-07-20:
I ended up having to use my HA server to flash it. 

I ran into the problem again when making a Esp32cam web server, so I wasn't able to use my HA server since I was using the Arduino IDE on my PC to flash it. 

The CH340 driver [here](https://sparks.gogo.co.nz/ch340.html) ended up fixing my issues.