adonno/tagreader #245: ESPHome 2023.7 - Error at firmware update

Eyco1981, 2023-07-19:
With ESPHome the following error occurs.
It seems, the configuration in your repository needs to be changed.

```INFO ESPHome 2023.7.0 INFO Reading configuration /config/esphome/tagreader-7c1293.yaml... Failed config switch.template: [source /config/esphome/.esphome/packages/fa8076cc/tagreader.yaml:56] platform: template name: TagReader Buzzer Enabled id: buzzer_enabled icon: mdi:volume-high optimistic: True The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead. restore_state: True entity_category: config switch.template: [source /config/esphome/.esphome/packages/fa8076cc/tagreader.yaml:63] platform: template name: TagReader LED enabled id: led_enabled icon: mdi:alarm-light-outline optimistic: True The restore_state option has been removed in 2023.7.0. Use the restore_mode option instead. restore_state: True entity_category: config
```

Masterz69, 2023-07-19:
> 

Remove ```restore_state``` from template switches.
Specify ```restore_mode``` if not defined yet.

jakevis, 2023-07-19:
yeah running into this issue as well - cant work out how to override the template - but I guess ideally this needs to be updated in https://github.com/adonno/tagreader/blob/master/tagreader.yaml

howesey2k, 2023-07-19:
I agree for it to be looked at. I've deleted the references to restore_state and reinstalled it. Seems to be working but getting an error "Component pn532 took a long time for an operation"
Could this be looked at? Many thanks

monstergubl, 2023-07-31:
hey got the same problem did you get a solution ?

waazdakka, 2023-08-19:
Hello,
I just flashed again my tagreader with the actual yaml but I still get the 
"Component pn532 took a long time for an operation (0.10 s).
[W][component:205]: Components should block for at most 20-30ms." error. 
Don't know what to change to make this error disappear :'(

howesey2k, 2023-08-19:
I get the same error. I think it's still working with this error but please
check too. I think someone mentioned in a forum post that it's possibly an
ESPHome issue. Might be worth mentioning it on their forums. The firmware
update fixed a different issue which stopped the device from working.

titis14, 2023-08-26:
Hi,
I still get the error even though I have the latest version.

howesey2k, 2023-08-26:
What error you getting exactly?

titis14, 2023-08-27:
This mistake.
[11:50:41][W][component:204]: Component pn532 took a long time for an operation (0.10 s).
[11:50:41][W][component:205]: Components should block for at most 20-30ms.
For a while now my tagreader has not been able to read badges. I made a topic here but no one replied.
https://github.com/adonno/tagreader/issues/241
I don't know if it's related to this error. I have this project and for several months at my club people can no longer deactivate the alarm easily.
Thanks

howesey2k, 2023-08-27:
Did you copy and paste the yaml update that adonno released? This fixed my
tags from reading. You need to copy and the paste the new yaml into the
correct file. Then click install again. This won't fix the timeout issue in
the logs but should fix the tags from not scanning. If this doesn't work,
try reaching out again and see if someone else can help you like a
developer :) If you are unsure where to copy and paste the yaml, I'll see
if I can remember haha

titis14, 2023-08-27:
On ESPHome I clicked on edited and I put the new yaml then installed by wifi.

howesey2k, 2023-08-27:
It's actually in a folder on your machine you need to navigate to. Do you
use a file editor plug in?

titis14, 2023-08-27:
No, I didn't do it through a file editor. On Home Assistant I use Studio Core Server, would you tell me which file to modify? Thanks

howesey2k, 2023-08-27:
I navigated to the yaml file by
esphome/.esphome/packages/name of tag reader package/Tagreader.yaml

I then copied Adonos code and pasted it into that yaml then clicked install
on the esphome integration

titis14, 2023-08-27:
Thank you in this way I am well in v1.4 :(

howesey2k, 2023-08-27:
Did this work?

titis14, 2023-08-27:
I think I know why it doesn't work anymore. Indeed I had to use the smart home junkie file.
dashboard_import:
   package_import_url: github://smarthomejunkie/tagreader/tagreader.yaml
In my editor I see that the file is not up to date.
I modified the file but when I want to validate it in ESPHome I have errors. I also have 2 folder for the tag.
Is there a way to erase everything and start from scratch?
I do all this remotely because I don't have the ESP near me.

howesey2k, 2023-08-27:
I was once given this link
https://adonno.github.io/tagreader

Plug it into the usb on your pc and use edge, chrome has never worked for
me.

If this doesn't work, not sure how to help sorry.

titis14, 2023-08-27:
Ok thanks you have already put on a track.
I'm going to do a new ESP and we'll see.

howesey2k, 2023-08-27:
Any joy?

closed 2023-10-15 by Eyco1981