adonno/tagreader #248: Compile of tagreader.yaml fails in ESPHome

arretx, 2023-07-31:
I'm not sure what is going on.  I've created a new tagreader.yaml file in /config/esphome/ and I've edited the SSID information for the WIFI, but when I go back to ESPHome in the UI and find the device and attempt to flash, or even download the flashable file, the compiler fails.

~~~
Compiling /data/tagreader-001/.pioenvs/tagreader-001/src/esphome/core/time.cpp.o
Compiling /data/tagreader-001/.pioenvs/tagreader-001/src/esphome/core/util.cpp.o
Compiling /data/tagreader-001/.pioenvs/tagreader-001/src/main.cpp.o
Generating LD script /data/tagreader-001/.pioenvs/tagreader-001/ld/local.eagle.app.v6.common.ld
Compiling /data/tagreader-001/.pioenvs/tagreader-001/lib905/ESPAsyncTCP-esphome/AsyncPrinter.cpp.o
xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/tagreader-001/.pioenvs/tagreader-001/src/main.cpp.o] Error 1
========================= [FAILED] Took 116.36 seconds =========================
~~~

No clue what to do.

qwik3r, 2023-10-15:
Did you ever figure this out? Seems like the original creator could care less.

adonno, 2024-06-20:
nothing about caring less. 

di you try with the original yaml an did it work would you mind pasting the changed section from your yaml ?