adonno/tagreader #250: Sandwich Version DIY Available?

misterjtc, 2023-08-16:
Hi,

I am wondering if there are any plans to opensource the design of the sandwich version which is sold on your website?

Thanks!

thicandido, 2023-10-16:
same here! :) waiting for this.

screenagerbe, 2023-11-13:
as do I 🤭

hundredfire, 2024-02-08:
Me four

SmartHomeFanatic, 2024-03-24:
You can find the case parts here:
https://www.printables.com/de/model/364699-tagreader-sandwith-version

Guess the rest remains the same.

pavlojs, 2024-04-04:
I have also prepared a PCB if someone prefer this way

#https://github.com/pavlojs/tagreader_pcb

RyanWor, 2025-03-27:
> I have also prepared a PCB if someone prefer this way
> 
> #https://github.com/pavlojs/tagreader_pcb

Thank you very much for doing and sharing this @pavlojs! Just assembled my first one today. 

[image]

[image]