adonno/tagreader #260: Not working reliable

Ayuus, 2024-05-19:
I have issues with reliability, both with my home-made version and the prebuilt version I bought. My cards are sometimes detected, but more orten than not, card reading results in rebooting. Same cards work flawlessly when scanned with my Android phone.
What can be the cause of this, and what can I do to fix this?

gdst4rp, 2024-06-01:
May be related to either of these two issues I've run across in the past:

- https://github.com/esphome/issues/issues/5208
     - NFC payload length being longer than 110 bytes causes an overflow. Test this with shorter length cards to see if it works. I had to re-write my NFC payloads to be shorter and essentially "unpack" them in NodeRed.

- https://github.com/adonno/tagreader/issues/261
     - More recent issue where the reader would reboot whenever I scanned a tag. Had to trim off blocks of unused code (at least for me) and re-install the yaml file.

Happy to assist with either one if you need more details. Not sure if this code is being maintained anymore sadly.

Ayuus, 2024-06-02:
Tried removing code: no luck. However, the [ESP32 version of the tagreader](https://github.com/leinich/tagreader) does work OK for me since several days (still testing).

closed 2024-07-11 by Ayuus

JohnMcLear, 2025-07-03:
@Ayuus  this sounds like https://github.com/adonno/tagreader/issues/222#issuecomment-3031950102