adonno/tagreader #261: ESPHome 2024.5.0 Breaks Tag Reader

gdst4rp, 2024-05-19:
With the update to 2024.5.0 simply scanning a tag that worked just fine in earlier versions, causes the the NFC tag reader to reboot.

Initial sound is played, LED light turns on, but nothing happens and Home Assistant doesn't register the tag scan event. The device then reboots and plays the bootup sound sequence.

[10:11:34][C][improv_serial:032]: Improv Serial:
[10:11:34][C][wifi_info:011]: WifiInfo SSID 'NFC Master Bathroom Connected SSID'
[10:11:34][C][wifi_info:011]:   Icon: 'mdi:wifi-strength-2'
[10:11:34][C][wifi_info:009]: WifiInfo IPAddress 'NFC Master Bathroom IP Address'
[10:11:34][C][wifi_info:009]:   Icon: 'mdi:wifi'
**[TAG SCANNED HERE]**
WARNING nfc-master-bathroom @ 192.168.XX.XX: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for nfc-master-bathroom @ 192.168.XX.XX
WARNING Disconnected from API
INFO Successfully connected to nfc-master-bathroom @ 192.168.XX.XX in 0.006s
INFO Successful handshake with nfc-master-bathroom @ 192.168.XX.XX in 0.027s

gdst4rp, 2024-05-25:
More details, and still broken as of ESPHome 2024.5.3. Here's what I get from the local USB connected logs when I scan a tag.

[09:04:56]
[09:04:56]User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

[09:04:56]Unhandled C++ exception: OOM
[09:04:56]
[09:04:56]>>>stack>>>
[09:04:56]
[09:04:56]ctx: cont
[09:04:56]sp: 3ffff890 end: 3fffffd0 offset: 0010
[09:04:56]3ffff8a0:  3fffaeb0 4020fc2b 000000ca 40233db4  
[09:04:56]3ffff8b0:  00000000 3fffaf2c 3fffb1ec 40245959  
[09:04:56]3ffff8c0:  000000c9 3fffaebc 3fffb19c 40210e98  
[09:04:56]3ffff8d0:  3fffaebc 3ffff8f0 3fffb19c 402112d9  
[09:04:56]3ffff8e0:  3fffaeb4 3fffb174 3fffb0ec 4020fe88  
[09:04:56]3ffff8f0:  3fffabc8 40245959 00000020 4010185c  
[09:04:56]3ffff900:  00000010 3f800000 3fffae5c 00000096  
[09:04:56]3ffff910:  00000000 40223940 00000020 4010185c  
[09:04:56]3ffff920:  00000014 3f800000 00000050 00000096  
[09:04:56]3ffff930:  3ffff9c4 3fff61dc 3fffb0e4 40223951  
[09:04:56]3ffff940:  40220c40 00000000 3fffae5c 3fffae63  
[09:04:56]3ffff950:  3fffae63 3fffae6c 00000010 0000001e  
[09:04:56]3ffff960:  3fff47d8 0000004e 00000074 3fffae94  
[09:04:56]3ffff970:  3fffb0ac 3fffb0c4 00000014 00000014  
[09:04:56]3ffff980:  3f800000 00000023 3f800000 3fff61dc  
[09:04:56]3ffff990:  40220c40 00000000 3fffa94c 3fffa953  
[09:04:56]3ffff9a0:  3fffa953 3fffa95c 00000010 0000001e  
[09:04:56]3ffff9b0:  3fff3c90 3fff47ac 3fff3c8c 3fffa984  
[09:04:56]3ffff9c0:  3fffab9c 3fffabb4 00000014 00000014  
[09:04:56]3ffff9d0:  3fffa738 4020fc2b 00000020 3fff61dc  
[09:04:56]3ffff9e0:  3fffabf4 3fffabfb 3fffabfb 3fffac04  
[09:04:56]3ffff9f0:  00000010 0000001e 00000020 4010185c  
[09:04:56]3ffffa00:  3fffa91c 3fffac2c 3fffae44 3fffabd4  
[09:04:56]3ffffa10:  00000014 00000014 3ffffa70 4020fef4  
[09:04:56]3ffffa20:  00000000 00000000 3fffa934 00000000  
[09:04:56]3ffffa30:  00000010 4020fc2b 3fffa45c 40233d9f  
[09:04:56]3ffffa40:  00000000 3fffa43c 3ffffa9c 40245959  
[09:04:56]3ffffa50:  00000014 4020fec3 00000020 40220508  
[09:04:56]3ffffa60:  3ffffad0 402238a8 3fff61dc 40220c7a  
[09:04:56]3ffffa70:  3fffa6e4 3fffa6eb 3fffa6eb 3fffa6f4  
[09:04:56]3ffffa80:  00000010 0000001e 3fffa6ac 4010185c  
[09:04:56]3ffffa90:  00000010 3fffa71c 3fffa934 3fffa6c4  
[09:04:56]3ffffaa0:  00000014 00000014 3ffffb28 40245959  
[09:04:56]3ffffab0:  00000014 3fff92cb 3fff92cb 40220508  
[09:04:56]3ffffac0:  3ffffc6c 3ffffb70 3fff420c 40220d3e  
[09:04:56]3ffffad0:  3fffa45c 3fffa463 3fffa463 3fffa46c  
[09:04:56]3ffffae0:  00000010 0000001e 00000010 3fff9d3c  
[09:04:56]3ffffaf0:  3fff9d3c 3fffa494 3fffa6ac 3fffa1d4  
[09:04:56]3ffffb00:  3fffa1db 3fffa1db 3fffa1e4 00000010  
[09:04:56]3ffffb10:  0000001e 3fff9fa4 3ffffb70 4020fef4  
[09:04:56]3ffffb20:  3fffa20c 3fffa424 3fffa43c 00000014  
[09:04:56]3ffffb30:  00000014 4020fc2b 3fff9ce4 40233d9f  
[09:04:56]3ffffb40:  3fff9f4c 00000014 00000014 40245959  
[09:04:56]3ffffb50:  00000014 4020fec3 00000020 40220508  
[09:04:56]3ffffb60:  3ffffc6c 3ffffbd0 3fff61bc 4022053c  
[09:04:56]3ffffb70:  3fff9f6c 3fff9f73 3fff9f73 3fff9f7c  
[09:04:56]3ffffb80:  00000010 0000001e 3fff9f34 4010185c  
[09:04:56]3ffffb90:  00000010 3fff9fa4 3fffa1bc 3ffffba4  
[09:04:56]3ffffba0:  00000000 00000000 3ffffbfc 40245959  
[09:04:56]3ffffbb0:  00000014 3fff923c 00000010 40220508  
[09:04:56]3ffffbc0:  3ffffc6c 3ffffc40 3fff61bc 40220da1  
[09:04:56]3ffffbd0:  3fff9ce4 3fff9ceb 3fff9ceb 3fff9cf4  
[09:04:56]3ffffbe0:  00000010 0000001e 3ffffc40 4020fef4  
[09:04:56]3ffffbf0:  00000000 3fff9d1c 3fff9f34 3fff9cc4  
[09:04:56]3ffffc00:  00000014 00000014 3fff93f4 40233d9f  
[09:04:56]3ffffc10:  00000000 3fff901c 3ffffc6c 40245959  
[09:04:56]3ffffc20:  00000014 4020fec3 00000020 40223ab4  
[09:04:56]3ffffc30:  3ffffca0 40220d68 3fff61bc 40220c7a  
[... 37 lines cut ...]
[09:04:57]3ffffe90:  3fff19f4 00000010 3fff802c 3fff79f4  
[09:04:57]3ffffea0:  3fff8884 00000014 00000014 3fff7dda  
[09:04:57]3ffffeb0:  3fff7de4 00000000 3fff85fc 00000014  
[09:04:57]3ffffec0:  00000014 3ffffed0 00000010 40214bbe  
[09:04:57]3ffffed0:  3fff7a0c 00000014 00000014 402130c8  
[09:04:57]3ffffee0:  00000080 3fffbff3 3fff7994 00000014  
[09:04:57]3ffffef0:  00000014 3fffff40 3fffff28 402131a0  
[09:04:57]3fffff00:  00000001 3fff41e4 3fff416c 40213386  
[09:04:57]3fffff10:  3fffdad0 3fff04a8 3fff04a8 4021dd17  
[09:04:57]3fffff20:  4026f026 3fff3944 3fffbfec 3fffbff3  
[09:04:57]3fffff30:  3fffbff3 3fffbebc 3fffbec9 3fffbed3  
[09:04:57]3fffff40:  3fff7dec 0002b7ee 07ef9db2 0324ab39  
[09:04:57]3fffff50:  3fffbec9 3fffbec2 3fff04a8 3fff416c  
[09:04:57]3fffff60:  3fff77b0 3fff416c 3fffff80 3fff416c  
[09:04:57]3fffff70:  3fff77b0 00000002 3fff04a8 4021adaa  
[09:04:57]3fffff80:  0002b7ee 3fff416c 3fff6ba4 3fff0cf0  
[09:04:57]3fffff90:  3fff77e0 3fff42dc 3fff04dc 3fff0cf0  
[09:04:57]3fffffa0:  3fffdad0 00000000 3fff0cc4 3fff0cf0  
[09:04:57]3fffffb0:  3fffdad0 00000000 3fff0cc4 40234368  
[09:04:57]3fffffc0:  feefeffe feefeffe 3fffdab0 40100561  
[09:04:57]<<<stack<<<
[09:04:57]
[09:04:57]last failed alloc call: 40245959(202)
[09:04:57]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

gdst4rp, 2024-05-26:
Even more info, stack trace from the HA server with local ESP8266 tag reader connected via USB.

[09:32:47]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[09:32:47]
[09:32:47]Unhandled C++ exception: OOM
[09:32:47]
[09:32:47]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[09:32:47]
[09:32:47]ctx: cont
[09:32:47]sp: 3ffff650 end: 3fffffd0 offset: 0010
[09:32:47]3ffff660:  000000c9 00000000 00000050 4022c394  
WARNING Decoded 0x4022c394: operator new(unsigned int)
[09:32:47]3ffff670:  3fffb0a4 3fffae34 3ffff6e0 4020f885  
WARNING Decoded 0x4020f885: esphome::nfc::NdefRecord::clone() const
[09:32:47]3ffff680:  3fffacd4 3fffaf3c 3ffff6e0 4020f560  
WARNING Decoded 0x4020f560: esphome::nfc::NfcTag::NfcTag(esphome::nfc::NfcTag const&)
[09:32:47]3ffff690:  00000000 4023aed1 00000020 4010178c  
WARNING Decoded 0x4023aed1: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data(char*) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:181
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:221
WARNING Decoded 0x4010178c: malloc
[09:32:47]3ffff6a0:  00000010 3fff0a44 3fffac7c 4022c37f  
WARNING Decoded 0x4022c37f: operator new(unsigned int)
[09:32:47]3ffff6b0:  00000000 3fffac5c 3ffff70c 4023aed1  
WARNING Decoded 0x4023aed1: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data(char*) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:181
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:221
[09:32:47]3ffff6c0:  00000014 4020f303 00000020 3fff5ce4  
WARNING Decoded 0x4020f303: std::__shared_ptr<esphome::nfc::NdefRecord, (__gnu_cxx::_Lock_policy)0>::__shared_ptr<esphome::nfc::NdefRecord, std::default_delete<esphome::nfc::NdefRecord>, void>(std::unique_ptr<esphome::nfc::NdefRecord, std::default_delete<esphome::nfc::NdefRecord> >&&)
[09:32:47]3ffff6d0:  3ffff78c 3ffff760 3fff5cf4 40224153  
WARNING Decoded 0x40224153: esphome::IfAction<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag>::play_complex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag)
[09:32:47]3ffff6e0:  3fffaf04 3fffaf0b 3fffaf0b 3fffaf14  
[09:32:47]3ffff6f0:  00000010 0000001e 00000010 4022c37f  
WARNING Decoded 0x4022c37f: operator new(unsigned int)
[09:32:47]3ffff700:  3fffaa4c 00000000 00000000 3fffaee4  
[09:32:47]3ffff710:  00000014 00000014 3fffaecc 4010178c  
WARNING Decoded 0x4010178c: malloc
[09:32:47]3ffff720:  00000010 4020f303 3fffa9f4 4022c37f  
WARNING Decoded 0x4020f303: std::__shared_ptr<esphome::nfc::NdefRecord, (__gnu_cxx::_Lock_policy)0>::__shared_ptr<esphome::nfc::NdefRecord, std::default_delete<esphome::nfc::NdefRecord>, void>(std::unique_ptr<esphome::nfc::NdefRecord, std::default_delete<esphome::nfc::NdefRecord> >&&)
WARNING Decoded 0x4022c37f: operator new(unsigned int)
[09:32:47]3ffff730:  40230aac 3fffa9d4 3ffff78c 4023aed1  
WARNING Decoded 0x40230aac: esphome::switch_::SwitchCondition<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag>::check(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag)
WARNING Decoded 0x4023aed1: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_data(char*) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h:181
 (inlined by) void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.tcc:221
[09:32:47]3ffff740:  00000014 4020f59b 00000020 4021f248  
WARNING Decoded 0x4020f59b: esphome::nfc::NfcTag::NfcTag(esphome::nfc::NfcTag const&)
WARNING Decoded 0x4021f248: esphome::LambdaAction<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag>::play(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag)
[09:32:47]3ffff750:  3ffff7c0 4022411c 3fff5cf4 4021f9ba  
WARNING Decoded 0x4022411c: esphome::IfAction<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag>::play_complex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag)
WARNING Decoded 0x4021f9ba: esphome::Action<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag>::play_next_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, esphome::nfc::NfcTag)
[09:32:47]3ffff760:  3fffac7c 3fffac83 3fffac83 3fffac8c  
[09:32:47]3ffff770:  00000010 0000001e 3fffac44 4010178c  
WARNING Decoded 0x4010178c: malloc
[09:32:47]3ffff780:  00000010 3fffacb4 3fffaecc 3fffac5c  
[09:32:47]3ffff790:  00000014 00000014 3ffff818 4023aed1  
[... cut ...]
[09:32:50]3fffffb0:  3fffdad0 00000000 3fff052c 4022c794  
WARNING Decoded 0x4022c794: loop_wrapper() at core_esp8266_main.cpp
[09:32:50]3fffffc0:  feefeffe feefeffe 3fffdab0 401004c9  
WARNING Decoded 0x401004c9: cont_wrapper
[09:32:50]<<<stack<<<
[09:32:50]
[09:32:50]last failed alloc call: 4020F885(80)
WARNING Memory allocation of 80 bytes failed at 4020F885
WARNING Decoded 0x4020f885: esphome::nfc::NdefRecord::clone() const
[09:32:50]
[09:32:50]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

gdst4rp, 2024-05-26:
I wasn't able to figure out what exactly caused the out of memory error, so I went through the tagreader.yaml file and removed components that I had no use for. **THIS MAY OR MAY NOT WORK YOU**, but I figured I'd document what I removed to get it work again. I only use this tagreader to send commands to HA directly, so I removed the ability to write tags, or recognize any non-HA tags. 

1.  Commented out the captive_portal

- captive_portal: --> **#**captive_portal:

2. Removed the entire section for "**Define buttons for writing tags via HA**"
3. Removed references to size_t for applemusic, spotify, and sonos
4. Removed the elseif sections that reference applemusic, spotify, and sonos
5. (not sure if this matters) Changed the 500ms timeouts to 200ms each. (2 instances of this change)

See below for changes:

# Insert your SSID and Your PWD after inital setup
wifi:
  ssid: !secret wifi_ssid          # Enter your WiFi SSID here. Example: `ssid: 'your_network_name'`
  password: !secret wifi_password      # Enter your wifi password here. Example: `password: 'abcde123456'`
  ap:
    ssid: ${name}

# Enable the captive portal for inital WiFi setup
**#**captive_portal:

dashboard_import:
  package_import_url: github://adonno/tagreader/tagreader.yaml

improv_serial:

substitutions:
  name: "nfc-guest-room"
  friendly_name: NFC Guest Room

esphome:
  name: $name
  platform: ESP8266
  board: d1_mini

  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: adonno.tag_reader
    version: "1.4"
# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"
    - light.turn_on:
        id: activity_led
        brightness: 100%
        red: 0%
        green: 0%
        blue: 100%
        flash_length: 500ms
    - switch.turn_on: buzzer_enabled
    - switch.turn_on: led_enabled

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
[... 295 lines cut ...]
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
      icon: mdi:wifi
      entity_category: diagnostic
    ssid:
      name: "${friendly_name} Connected SSID"
      icon: mdi:wifi-strength-2
      entity_category: diagnostic

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

# Configure LED
light:
- platform: neopixelbus
  variant: WS2812
  pin: D8
  num_leds: 1
  flash_transition_length: 500ms
  type: GRB
  id: activity_led
  name: "${friendly_name} LED"
  restore_mode: ALWAYS_OFF

gdst4rp, 2024-05-26:
I didn't figure out how to fix the code itself, but since it was an Out of Memory error, I removed the following sections since I don't need them in my use case. I only scan HA tags, and never write tags from this device.

1. Comment out the "captive_portal:" line.
2. Delete the entire section for writing tags

- From the line "Define buttons for writing tags via HA" all the way to the line above "Enable logging"
3. Delete the 3 lines referencing size_t for applemusic, spotify, and sonos:

- size_t applemusic = payload.find("https://music.apple.com");
- size_t spotify = payload.find("https://open.spotify.com");
- size_t sonos = payload.find("sonos-2://");

4. Delete the elseif sections for applemusic, spotify, and sonos

- `  else if (type == "U" and applemusic != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Apple Music tag NDEF");
              id(source)="amusic";
              id(url)=payload;
            }
            else if (type == "U" and spotify != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Spotify tag NDEF");
              id(source)="spotify";
              id(url)=payload;
            }
            else if (type == "U" and sonos != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Sonos app tag NDEF");
              id(source)="sonos";
              id(url)=payload;
            }`

5. (Not sure if this matters) Change the two references to 500ms delay to 200ms

**ORIGINAL TAGREADER.YAML I WAS USING PRIOR TO CHANGES AS REFERENCE**

`# Insert your SSID and Your PWD after inital setup
wifi:
  ssid: !secret wifi_ssid          # Enter your WiFi SSID here. Example: `ssid: 'your_network_name'`
  password: !secret wifi_password      # Enter your wifi password here. Example: `password: 'abcde123456'`
  ap:
    ssid: ${name}

# Enable the captive portal for inital WiFi setup
captive_portal:

dashboard_import:
  package_import_url: github://adonno/tagreader/tagreader.yaml

improv_serial:

substitutions:
  name: "nfc-guest-room"
  friendly_name: NFC Guest Room

esphome:
  name: $name
  platform: ESP8266
  board: d1_mini

  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: adonno.tag_reader
    version: "1.4"
# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
[... 317 lines cut ...]
    ip_address:
      name: "${friendly_name} IP Address"
      icon: mdi:wifi
      entity_category: diagnostic
    ssid:
      name: "${friendly_name} Connected SSID"
      icon: mdi:wifi-strength-2
      entity_category: diagnostic

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

# Configure LED
light:
- platform: neopixelbus
  variant: WS2812
  pin: D8
  num_leds: 1
  flash_transition_length: 500ms
  type: GRB
  id: activity_led
  name: "${friendly_name} LED"
  restore_mode: ALWAYS_OFF
`

gdst4rp, 2024-06-01:
Not sure if this codebase is being maintained, but I got it to work by stripping out blocks of code that were not needed.

1. Comment out the "captive_portal:" line.
2. Delete the entire section for writing tags
- From the line "Define buttons for writing tags via HA" all the way to the line above "Enable logging"

3. Delete the 3 lines referencing size_t for applemusic, spotify, and sonos:
- size_t applemusic = payload.find("https://music.apple.com");
- size_t spotify = payload.find("https://open.spotify.com");
- size_t sonos = payload.find("sonos-2://");

4. Delete the elseif sections for applemusic, spotify, and sonos
-```
 `  else if (type == "U" and applemusic != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Apple Music tag NDEF");
              id(source)="amusic";
              id(url)=payload;
            }
            else if (type == "U" and spotify != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Spotify tag NDEF");
              id(source)="spotify";
              id(url)=payload;
            }
            else if (type == "U" and sonos != std::string::npos ) {
              ESP_LOGD("tagreader", "Found Sonos app tag NDEF");
              id(source)="sonos";
              id(url)=payload;
            }`
```

5. (Not sure if this matters) Change the two references to 500ms delay to 200ms

imhotep, 2024-06-05:
@gdst4rp does your workaround still work? Is it stable? I am tempted to just install an esphome version from last year as this repository does not seem to be maintained.

imhotep, 2024-06-06:
@gdst4rp do you mind sharing your full YAML file here?

gdst4rp, 2024-06-06:
[updated-tagreader.txt](https://github.com/user-attachments/files/15642862/updated-tagreader.txt)

Here you go. Change up the substitutions section to update the device name to what you want. I also use the separate !secrets files for SSID and SSID Password values.

adonno, 2024-06-20:
captive portal is still required for first time users, so i cannot comment it out 
i need to test it further. but i have to admit i am not yet sure bout removing functionalities  
I wonder why it suddenly start rebooting

gdst4rp, 2024-06-20:
From what I can tell, it's a memory overflow on the 8266 chips. Are the crash logs above helpful at all? I don't know how to parse them so I just tried to trim some functionality, which seemed to work.

adonno, 2024-06-20:
I wonder if i should write a light version for people facing issues.. MAybe one without the jukebox component etc. 
I'va had some more people complaining that indeed teh reader keeps reseting after reading a tag. And they are getting worse. 

Maybe people will be willing to have less functionality for a working device.

gdst4rp, 2024-06-20:
For me personally, all I use it for is to scan HA tags. I have a whole flow in NodeRed that parses the tags, calls the right speakers to play from and plays from the right Spotify source. I don't use the tagreader to actually write new tags, or call directly to Apple, YouTube, or Spotify, which is why I attempted to cut out all the extra code, but I'm sure I missed some things.

I've already had to shorten the tag payload per that other issue since it was crashing if the payload was too big.

imhotep, 2024-06-23:
> I wonder if i should write a light version for people facing issues.. MAybe one without the jukebox component etc.
I'va had some more people complaining that indeed teh reader keeps reseting after reading a tag. And they are getting worse.
Maybe people will be willing to have less functionality for a working device.

Yes, please! I just want to be able to scan tags. Absolutely nothing else. I haven't been able to use the device with the current config.

SaisonG, 2024-07-01:
> > I wonder if i should write a light version for people facing issues.. MAybe one without the jukebox component etc.
> > I'va had some more people complaining that indeed teh reader keeps reseting after reading a tag. And they are getting worse.
> > Maybe people will be willing to have less functionality for a working device.
> 
> @adonno Yes, please! I just want to be able to scan tags. Absolutely nothing else. I haven't been able to use the device with the current config.

Same for me.  Same behavior, can't get the tag reader working. Just need it to scan tags.

hitnrun30, 2024-07-17:
I might be a bit off since I am using a esp32 wemos D1Mini, but after I added 

power_save_mode: none

to the wifi of my esphome yaml, all was relatively good.  I am still getting some warnings and an error but the sensor works.

Im trying to get rid of the below errors and warnings

[E][pn532.mifare_classic:101]: Authentication failed - Block 0x04
[W][component:237]: Component pn532 took a long time for an operation (109 ms).
[W][component:238]: Components should block for at most 30 ms.

imhotep, 2024-07-28:
I spent some time messing around with the config today and for some reason the lambda makes the device crash 100% of the time on tag scanned. I simplified it a LOT and now it works 100% of the time. Here's my config

<details>
  <summary>tagreader.yaml</summary>

  ```yml
# Insert your SSID and Your PWD after inital setup
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${name}

# Enable the captive portal for inital WiFi setup
captive_portal:

substitutions:
  name: tagreader
  friendly_name: TagReader

improv_serial:

esphome:
  name: $name
  platform: ESP8266
  board: d1_mini

  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: false

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: adonno.tag_reader
    version: "1.4"
  # If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"
    - light.turn_on:
        id: activity_led
        brightness: 100%
        red: 0%
        green: 0%
        blue: 100%
        flash_length: 500ms
    - switch.turn_on: buzzer_enabled
    - switch.turn_on: led_enabled

# Define buttons for writing tags via HA 
button:
  - platform: template
    name: Write Tag Random
    id: write_tag_random
    # Optional variables:
    icon: "mdi:pencil-box"
    on_press:
      then:
      - light.turn_on:
          id: activity_led
          brightness: 100%
          red: 100%
          green: 0%
          blue: 100%    
      - lambda: |-
[... 186 lines cut ...]
      icon: mdi:wifi
      entity_category: diagnostic
    ssid:
      name: "${friendly_name} Connected SSID"
      icon: mdi:wifi-strength-2
      entity_category: diagnostic

# Configure LED
light:
  - platform: neopixelbus
    variant: WS2812
    pin: D8
    num_leds: 1
    flash_transition_length: 500ms
    type: GRB
    id: activity_led
    name: "${friendly_name} LED"
    restore_mode: ALWAYS_OFF

# Enable OTA upgrade
ota:
  platform: esphome
  ```

</details>

FastHowTo, 2024-12-02:
The code above provided by @imhotep doesn't work.  When a tag is scanned, it returns "Tag " + the serial number of the tag to home assistant, instead of the tag_id uri.  The result of this is a new tag being registered in Home Assistant, but when I scan that same tag with my phone, the correct tag is scanned.  When I scan it with the scanner, the new "Tag <serial number>" entity registers that it was scanned.

gdst4rp, 2024-12-02:
I'm still using the modified version that I posted back on June 1, and it seems to work just fine so far. See if that works for you.

https://github.com/adonno/tagreader/issues/261#issuecomment-2143568645

Youddha, 2025-01-17:
Hello!
Any news on this reboot/crash issue ?
Is there any version of an ESP where the original code is working ?

imhotep, 2025-01-17:
@Youddha have you tried any of the workarounds posted in this thread? A "crash/reboot" can be due to various reasons. Perhaps post a log of the crash if it's different from the one in this issue?

@FastHowTo change the lambda to make it return what you want? What I posed above aligns with what ESPHome [recommends](https://esphome.io/components/binary_sensor/pn532.html#ndef-reading) so I am not sure what you mean by "doesn't work".

FastHowTo, 2025-01-17:
I gave up on it due to the aforementioned issues and haven't picked it back up again since.

scandrive, 2025-01-17:
@Youddha 

I´ve just started with this project an ordered the parts mentioned a few weeks ago and assembled them last week. 
For my part I can say, initial it works with the original yaml in HA 2025.1.x at least for reading. To get it in the actual ESPhome Builder I have had to make some adjustments, but now it works also.
What I haven´t try yet is updating via ESPHome Builder.

gdst4rp, 2025-01-18:
Just chiming in to say that my devices are still working (on HA version 2024.12.5 and ESPHome version 2024.12.2). I haven't jumped on the latest versions of either yet since I ran into some snags (unrelated) with Zigbee that caused me to roll back.

I'm using the version of the code I linked above and all seems to work just fine.

Youddha, 2025-01-19:
On my side my tagreader is still rebooting after some (all?) tag scans. I started to look into TagTuner, a similar (forkee?) project from a contributor of the tagreader.
The firmware is great but the physical case is not as compact as adonno's one. Nonethess might be a solution for those with this problem too.

luka6000, 2025-03-10:
check this #293