adonno/tagreader #267: Sensor doesn't read tag

abdullahvenom21, 2024-07-12:
I build the tag and in home assistant it reacts to clean tag,cancel tag,write random tag but does not react to any card when tapped against the tag module i tried with the tags that came with and also with a credit card but it does not respond to any card and at same time my phone picks up the cards[image]

tim-vk, 2024-07-30:
Sorry for highjacking the post but I have exactly the same issue (but maybe a different cause). But lets run through the checklist together (it hasn't helped me (yet) but maybe you're an excellent rubber duck ;) ).

wires: 
- SCL to pin D1 - check
- SDA to pin D2 - check 
- buzzer to D7 - check
- Led to D8 -> missing for you but shouldn't matter

switches:
switch 1: ON - check
switch 2: OFF - check

Firmware on device: Go to homeassistant & check if the tagreader is there.

Power on behavior:

- buzzer should beep
- power LED on the reader
- LED will shortly blink

Tags: 
- Does it work with something else? (e.g. phone?) - check

Check serial logs:

Connect to the device & check the logs on boot. In my case it says:

```
[14:23:17][C][pn532:431]: PN532:
[14:23:17][C][pn532:443]:   Update Interval: 1.0s
[14:23:17][C][pn532_i2c:125]:   Address: 0x24
[14:23:17][E][component:082]:   Component pn532 is marked FAILED
```

Which is a bummer since that (for me) is the reason it won't read/write. The solutions I can find is make sure the switches are correct (which i think they are) but maybe there are other possible causes for this.

Lemme know if this helps you or if you have some additions.

abdullahvenom21, 2024-07-30:
Thank you for the help and i have the led but as the system is not working yet so i figured i will add them when the system works.
Right now i am making virtal home assitant on my pc and ill try to get the versions recomended for the tag reader i hope it works
And this is my first ever esp home project ill try my best to be rubber duck.

tim-vk, 2024-07-30:
Ah, well then lets start at the beginning: Did you get esphome installed on the tag reader?

abdullahvenom21, 2024-07-30:
Yes! I guess so,
I copied the file in the tag reader article on github and pasted it while editing the tag reader file.
Then i flashed my D1-mini with the same file with no changes.
Then the D1-mini created its own hotspot and connected the device to my home internet from there.
Then i connected the device to my home assistant server via cable
It started showing on the esp home dashboard.
Then i got the api key from the esp home dashboard and found D1-mini's ip address from my roter.
Then i connected the D1-mini to my home assistant server via the esp hone integration.
Then it only dives me option to:
* cancel Tag
* write tag
* another option
I don't really remember it also gave me a restart button on home assistant dashboard the buzzer works when ever i change the tag to canel tag or write tag but never works when i bring a tag near but it should work😓

tim-vk, 2024-07-30:
Awesome!

So If you enter the IP of your device you should be able to see the logging somewhere. Check the log & restart the device. If you can copy all the logging it writes when booting here that should be a good help.

jes1417, 2024-11-12:
Sorry to hijack, I was just informed that our device is no longer reading tags. I am not sure how long this has been an issue but here are the logs after a restart. This has been working for over a year and no changes have been made other than updating HA and ESPHome.

```
INFO ESPHome 2024.10.3
INFO Reading configuration /config/alarm-tag-reader-9bfe6f.yaml...
INFO Starting log output from 10.30.46.84 using esphome API
INFO Successfully connected to alarm-tag-reader-9bfe6f @ 10.30.46.84 in 0.010s
INFO Successful handshake with alarm-tag-reader-9bfe6f @ 10.30.46.84 in 0.037s
[15:56:09][I][app:100]: ESPHome version 2024.10.3 compiled on Nov 12 2024, 11:00:27
[15:56:09][I][app:102]: Project adonno.tag_reader version 1.4
[15:56:09][C][wifi:600]: WiFi:
[15:56:09][C][wifi:428]:   Local MAC: E8:DB:84:9B:FE:6F
[15:56:09][C][wifi:433]:   SSID: [redacted]
[15:56:09][C][wifi:436]:   IP Address: 10.30.46.84
[15:56:09][C][wifi:439]:   BSSID: [redacted]
[15:56:09][C][wifi:441]:   Hostname: 'alarm-tag-reader-9bfe6f'
[15:56:09][C][wifi:443]:   Signal strength: -59 dB ▂▄▆█
[15:56:09][C][wifi:447]:   Channel: 1
[15:56:09][C][wifi:448]:   Subnet: 255.255.255.0
[15:56:09][C][wifi:449]:   Gateway: 10.30.46.1
[15:56:09][C][wifi:450]:   DNS1: 10.30.46.201
[15:56:09][C][wifi:451]:   DNS2: 1.1.1.1
[15:56:09][C][logger:185]: Logger:
[15:56:09][C][logger:186]:   Level: DEBUG
[15:56:09][C][logger:188]:   Log Baud Rate: 115200
[15:56:09][C][logger:189]:   Hardware UART: UART0
[15:56:09][C][i2c.arduino:071]: I2C Bus:
[15:56:09][C][i2c.arduino:072]:   SDA Pin: GPIO4
[15:56:09][C][i2c.arduino:073]:   SCL Pin: GPIO5
[15:56:09][C][i2c.arduino:074]:   Frequency: 400000 Hz
[15:56:09][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[15:56:09][C][esp8266_pwm:022]: ESP8266 PWM:
[15:56:09][C][esp8266_pwm:023]:   Pin: GPIO13
[15:56:09][C][esp8266_pwm:024]:   Frequency: 988.0 Hz
[15:56:09][C][light:103]: Light 'TagReader LED'
[15:56:09][C][light:105]:   Default Transition Length: 1.0s
[15:56:09][C][light:106]:   Gamma Correct: 2.80
[15:56:09][C][template.switch:068]: Template Switch 'TagReader Buzzer Enabled'
[15:56:09][C][template.switch:070]:   Icon: 'mdi:volume-high'
[15:56:09][C][template.switch:090]:   Restore Mode: restore defaults to ON
[15:56:09][C][template.switch:057]:   Optimistic: YES
[15:56:10][C][template.switch:068]: Template Switch 'TagReader LED enabled'
[15:56:10][C][template.switch:070]:   Icon: 'mdi:alarm-light-outline'
[15:56:10][C][template.switch:090]:   Restore Mode: restore defaults to ON
[15:56:10][C][template.switch:057]:   Optimistic: YES
[15:56:10][C][restart.button:017]: Restart Button 'TagReader Restart'
[15:56:10][C][restart.button:017]:   Icon: 'mdi:restart'
[15:56:10][C][pn532:431]: PN532:
[15:56:10][C][pn532:443]:   Update Interval: 1.0s
[15:56:10][C][pn532_i2c:125]:   Address: 0x24
[15:56:10][C][status:034]: Status Binary Sensor 'TagReader Status'
[15:56:10][C][status:034]:   Device Class: 'connectivity'
[15:56:10][C][version.text_sensor:021]: Version Text Sensor 'TagReader ESPHome Version'
[15:56:10][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[15:56:10][C][rtttl:029]: Rtttl
[15:56:10][C][captive_portal:089]: Captive Portal:
[15:56:10][C][mdns:116]: mDNS:
[15:56:10][C][mdns:117]:   Hostname: alarm-tag-reader-9bfe6f
[15:56:10][C][esphome.ota:073]: Over-The-Air updates:
[15:56:10][C][esphome.ota:074]:   Address: alarm-tag-reader-9bfe6f.local:8266
[15:56:10][C][esphome.ota:075]:   Version: 2
[15:56:10][C][safe_mode:018]: Safe Mode:
[15:56:10][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:56:10][C][safe_mode:021]:   Invoke after 10 boot attempts
[15:56:10][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[15:56:10][C][api:140]: API Server:
[15:56:10][C][api:141]:   Address: alarm-tag-reader-9bfe6f.local:6053
[15:56:10][C][api:145]:   Using noise encryption: NO
[15:56:10][C][improv_serial:032]: Improv Serial:
[15:56:10][C][wifi_info:012]: WifiInfo SSID 'TagReader Connected SSID'
[... 31 lines cut ...]
[15:56:22][C][version.text_sensor:021]: Version Text Sensor 'TagReader ESPHome Version'
[15:56:22][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[15:56:22][C][rtttl:029]: Rtttl
[15:56:22][C][captive_portal:089]: Captive Portal:
[15:56:22][C][mdns:116]: mDNS:
[15:56:22][C][mdns:117]:   Hostname: alarm-tag-reader-9bfe6f
[15:56:22][C][esphome.ota:073]: Over-The-Air updates:
[15:56:22][C][esphome.ota:074]:   Address: alarm-tag-reader-9bfe6f.local:8266
[15:56:22][C][esphome.ota:075]:   Version: 2
[15:56:22][D][rtttl:387]: State changed from STATE_RUNNING to STATE_STOPPED
[15:56:22][D][rtttl:364]: Playback finished
[15:56:22][C][safe_mode:018]: Safe Mode:
[15:56:22][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:56:22][C][safe_mode:021]:   Invoke after 10 boot attempts
[15:56:22][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[15:56:22][C][api:140]: API Server:
[15:56:22][C][api:141]:   Address: alarm-tag-reader-9bfe6f.local:6053
[15:56:22][C][api:145]:   Using noise encryption: NO
[15:56:23][C][improv_serial:032]: Improv Serial:
[15:56:23][C][wifi_info:012]: WifiInfo SSID 'TagReader Connected SSID'
[15:56:23][C][wifi_info:012]:   Icon: 'mdi:wifi-strength-2'
[15:56:23][C][wifi_info:010]: WifiInfo IPAddress 'TagReader IP Address'
[15:56:23][C][wifi_info:010]:   Icon: 'mdi:wifi'
[15:57:14][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```