adonno/tagreader #268: Code for normal LED

Xa4-J, 2024-07-14:
If I wan't to use a normal LED incase of a RGB LED. What would be the code?

tim-vk, 2024-07-30:
There are 3 things you can do:

- The easiest way: Don't use the led at all.
- The easy way (that will probably still work): Just connect the + pin of your led to D8. It will probably blink a bit funny but my guess it will still work.
- The harder way (that will definitely work): Rewrite [tagreader.yaml](https://github.com/adonno/tagreader/blob/master/tagreader.yaml) to use a [binary light](https://esphome.io/components/light/binary) instead of the neopixel one that is currently used.

Xa4-J, 2024-08-02:
It worked by connecting a LED via a transistor to D8 with the code below. There will still be a lot of code that's not neccesary but it works.

`# Insert your SSID and Your PWD after inital setup
wifi:
  ssid: ""
  password: ""
  power_save_mode: none
  fast_connect: true
  
# Optional manual IP
  manual_ip:
    static_ip: 192.168.13.38
    gateway: 192.168.13.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tagreader Fallback Hotspot"
    password: ""

improv_serial:

substitutions:
  name: tagreader
  friendly_name: TagReader

esphome:
  name: $name
  platform: ESP8266
  board: d1_mini

  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: true

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: adonno.tag_reader
    version: "1.4"
# If buzzer is enabled, notify on api connection success
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!
    - rtttl.play: "success:d=24,o=5,b=100:c,g,b"
    - output.turn_on:
        id: activity_led
    - switch.turn_on: buzzer_enabled
    - switch.turn_on: led_enabled
    - delay: 10s
    - output.turn_on:
        id: activity_led

# Define switches to control LED and buzzer from HA
switch:
- platform: template
  name: "${friendly_name} Buzzer Enabled"
  id: buzzer_enabled
  icon: mdi:volume-high
  optimistic: true
  restore_mode: RESTORE_DEFAULT_ON
  entity_category: config
- platform: template
  name: "${friendly_name} LED enabled"
  id: led_enabled
  icon: mdi:alarm-light-outline
  optimistic: true
[... 303 lines cut ...]
binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
    entity_category: diagnostic

text_sensor:
  - platform: version
    hide_timestamp: true
    name: "${friendly_name} ESPHome Version"
    entity_category: diagnostic
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
      icon: mdi:wifi
      entity_category: diagnostic
    ssid:
      name: "${friendly_name} Connected SSID"
      icon: mdi:wifi-strength-2
      entity_category: diagnostic

# Define buzzer as output for RTTTL
rtttl:
  output: buzzer

`

closed 2024-08-02 by Xa4-J