adonno/tagreader #280: Case/STL file for Wemos D1 with USB-C

ilpantera, 2025-01-29:
Hi, 
would be great to add an Wemos D1 with USB-C option for the case. Has anyone already build one that could be uploaded here?
Thanks!

htims1989, 2025-02-08:
I designed (followed YouTube tutorials for blender 😉) an alternative case that a D1 USB-C board fits into, stacks the boards (the NFC board clips onto the lid) and has a round cutout with a thinner wall to allow the LED to shine through.

Could do with a little more work to make the lid clip on/off, but I just glued it and moved on! Included a zip with the STLs for the body/lid and a 3mf that I used to print using my Bambu A1. Let me know if you use it ☺️

[image]

[image]

[image]

[image]

[TagReaderCase.zip](https://github.com/user-attachments/files/18717610/TagReaderCase.zip)

quack3d, 2025-02-16:
I modified the existing one for USB-C version (D1 Mini V4.0.0) and added a few minor tweaks.

[image]

[Tag reader case USB-C.zip](https://github.com/user-attachments/files/18814782/Tag.reader.case.USB-C.zip)

quack3d, 2025-02-22:
Now I modified this Sandwich model which I prefer instead. https://www.printables.com/model/864845-tagreader-sandwich-hand-soldered . I moved the port for USB-C a bit, added a pole for the ESP to rest on and made the diameter of the tag reader attachment a tiny bit smaller.

[image]

[image]

[Tag reader case USB-C - sandwich.zip](https://github.com/user-attachments/files/18926004/Tag.reader.case.USB-C.-.sandwich.zip)

pavlojs, 2025-04-04:
[This PCB and case](https://github.com/pavlojs/tagreader_pcb) should fit USB-C connector

RyanWor, 2025-04-04:
Here is the modified bottom STL for the USB-C D1 Minis I bought here: https://www.amazon.com/dp/B0BR9GBNZH

[bottom.zip](https://github.com/user-attachments/files/19605388/bottom.zip)
(Zipped because GitHub would not allow direct upload of STL file)

Here is image of my completed PCB for visual context. 
[image]

Happy to add this as an issue/pull request to your repo if you like@pavlojs

pavlojs, 2025-04-11:
> Here is the modified bottom STL for the USB-C D1 Minis I bought here: https://www.amazon.com/dp/B0BR9GBNZH
>
> [bottom.zip](https://github.com/user-attachments/files/19605388/bottom.zip) (Zipped [...]

Sure. USB-C case is nice addition.