adonno/tagreader #285: wrong i2c address?

pdrobek, 2025-02-11:
Hi

I am struggling with esphome and pn532. My wemos d1 mini found i2c device on address 0x28 while yaml seeks some device on 0x24. I guess it is main issue why my pn532 was reported as Component pn532 is marked FAILED

Does anyone know how change i2c address, or show different way to resolve my issues?

[20:50:44][I][i2c.arduino:096]: Results from i2c bus scan:
**[20:50:44][I][i2c.arduino:102]: Found i2c device at address 0x28**
[20:50:44][C][esp8266_pwm:022]: ESP8266 PWM:
[20:50:44][C][esp8266_pwm:023]:   Pin: GPIO13
[20:50:45][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[20:50:45][C][light:092]: Light 'TagReader LED'
[20:50:45][C][light:094]:   Default Transition Length: 1.0s
[20:50:45][C][light:095]:   Gamma Correct: 2.80
[20:50:45][C][template.switch:068]: Template Switch 'TagReader Buzzer Enabled'
[20:50:45][C][template.switch:070]:   Icon: 'mdi:volume-high'
[20:50:45][C][template.switch:090]:   Restore Mode: restore defaults to ON
[20:50:45][C][template.switch:057]:   Optimistic: YES
[20:50:45][C][template.switch:068]: Template Switch 'TagReader LED enabled'
[20:50:45][C][template.switch:070]:   Icon: 'mdi:alarm-light-outline'
[20:50:45][C][template.switch:090]:   Restore Mode: restore defaults to ON
[20:50:45][C][template.switch:057]:   Optimistic: YES
[20:50:45][C][restart.button:017]: Restart Button 'TagReader Restart'
[20:50:45][C][restart.button:017]:   Icon: 'mdi:restart'
[20:50:45][C][pn532:431]: PN532:
[20:50:45][C][pn532:443]:   Update Interval: 1.0s
**[20:50:45][C][pn532_i2c:125]:   Address: 0x24
[20:50:45][E][component:082]:   Component pn532 is marked FAILED**

pdrobek, 2025-02-11:
after change address to 0x28 :/

i2c:
  scan: True
  frequency: 400kHz
  sda: D5
  scl: D6

globals:
  - id: source
    type: std::string
  - id: url
    type: std::string
  - id: info
    type: std::string
 
**pn532_i2c:
  address: 0x28**

[21:17:33][C][i2c.arduino:072]:   SDA Pin: GPIO14
[21:17:33][C][i2c.arduino:073]:   SCL Pin: GPIO12
[21:17:33][C][i2c.arduino:074]:   Frequency: 400000 Hz
[21:17:33][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[21:17:33][I][i2c.arduino:096]: Results from i2c bus scan:
[21:17:33][I][i2c.arduino:102]: Found i2c device at address 0x28
[21:17:33][C][esp8266_pwm:022]: ESP8266 PWM:
[21:17:33][C][esp8266_pwm:023]:   Pin: GPIO13
[21:17:33][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[21:17:33][C][light:092]: Light 'TagReader LED'
[21:17:33][C][light:094]:   Default Transition Length: 1.0s
[21:17:33][C][light:095]:   Gamma Correct: 2.80
[21:17:33][C][template.switch:068]: Template Switch 'TagReader Buzzer Enabled'
[21:17:33][C][template.switch:070]:   Icon: 'mdi:volume-high'
[21:17:33][C][template.switch:090]:   Restore Mode: restore defaults to ON
[21:17:33][C][template.switch:057]:   Optimistic: YES
[21:17:33][C][template.switch:068]: Template Switch 'TagReader LED enabled'
[21:17:33][C][template.switch:070]:   Icon: 'mdi:alarm-light-outline'
[21:17:33][C][template.switch:090]:   Restore Mode: restore defaults to ON
[21:17:33][C][template.switch:057]:   Optimistic: YES
[21:17:33][C][restart.button:017]: Restart Button 'TagReader Restart'
[21:17:33][C][restart.button:017]:   Icon: 'mdi:restart'
[21:17:33][C][pn532:431]: PN532:
[21:17:33][C][pn532:443]:   Update Interval: 1.0s
[21:17:33][C][pn532_i2c:125]:   Address: 0x28
[21:17:33][E][component:082]:   Component pn532 is marked FAILED

stablestud, 2025-06-21:
I had the same issue, I forgot to change the switches on the pn532 board.

From the README.md
>Also, make sure that you have set the switches on the PN532 to the following:
>
>    Switch 1: On (up)
>    Switch 2: Off (down)
>
>This enables the PN532 module to communicate with the D1 over I2C, and is required for the modules to work together!

Have you set the switches correctly?